Ultimaker/Cura #1225: Cannot turn off M109 on extruder change

karabas, 2016-12-07:
Sometimes due to temperature fluctuations it lowes down a bit, f. e. 208/210
and nozzle goes to tower and waits 210 exactly, melting plastic on the corner of the tower.
This causes nozzle strike on next movements and is dangerous and unnecessary.
Any ways to turn off M109 or how can I edit extruder change gcodes like in cura 15.04?

maukcc, 2016-12-20:
I would like to see an "extruder change gcode" as well.

nallath, 2016-12-20:
@BagelOrb: Could you chime in on this?

karabas, 2016-12-26:
> I would like to see an "extruder change gcode" as well.
You can write it in extruder files. Similar to fdmextruder.
But some Gcodes are hardcoded and  cannot be removed.

Ghostkeeper, 2016-12-27:
We have those. We call the settings `machine_extruder_start_code` and `machine_extruder_end_code`. They are in `fdmextruder.def.json`, so they should be in the extruder definition files.

maukcc, 2016-12-29:
Ahh, very nice. Is it also possible to add variables to them, as It does not seem to populate the variables?
Like
M109 S{material_standby_temperature}

It would also be very nice to be able to change this from within the GUI

Ghostkeeper, 2016-12-29:
No, it isn't. I added that for you real quick. https://github.com/Ultimaker/Cura/commit/341b92860c00ff4cd901a67fe89593f5689906ec

That won't be in the next release though.

maukcc, 2017-01-08:
I also tested this, and it does not seem to populate the variables. So the code ends up like:
;end extruder0 change
M104 S{material_standby_temperature}
G1 X70 Y20 F9000

Drayson80, 2017-01-09:
thanks for confirming... so it´s not my weired system :-)

BagelOrb, 2017-01-12:
The reason that temperature commands cannot be controlled by editing the extruder start gcode (etc.) is that the temperature commands are inserted some time before the extruder change.

That functionality is in place for the start gcode and the end gcode, though, but it isn't ported to the extruder start and extruder end.

It shouldn't be too difficult to port it, but that's not on our priority list. You are welcome to submit a pull request for CuraEngine. We can then merge your code fixes so that the feature will become part of the next version of Cura.

maukcc, 2017-01-12:
:) 
I was just making a new issue on this

ianpaschal, 2018-02-21:
Since this hasn't been touched in over a year, I'm closing this issue.

closed 2018-02-21 by ianpaschal