Ultimaker/Cura #1227: cannot change object position using 'Move' UI

mikosik, 2016-12-08:
It is not possible to change object position using Move UI ...

[image]

.. if object position has been changed by dragging object using mouse/touchpad.
Move UI works correctly when object has been just loaded from stl file and have been only clicked once to bring 'Move' UI.

Reproduced in cura 2.3.1 on Linux Mint 18.

Ghostkeeper, 2016-12-08:
Can't reproduce this in version 2.4 (master).

Note that you have to press enter or remove the focus from the text box before the move gets applied. Otherwise we'd have to update the position while typing, which gets annoying with large models.

mikosik, 2017-03-22:
@Ghostkeeper I confirm that it works correctly with cura 2.4.0 for linux official release.

closed 2017-03-22 by mikosik