Ultimaker/Cura #1229: Skirt intersects part (incorrect built skirt)

kotoezh, 2016-12-08:
On certain geometries skirt is built incorrect and intersects bottom layer:
[image]

Low Quality (0.15mm) profile and Skirt should be set to reproduce the bug

STL model included:
[proto1_mount_socket.zip](https://github.com/Ultimaker/Cura/files/639205/proto1_mount_socket.zip)

Ghostkeeper, 2016-12-08:
Can't reproduce this in Cura 2.4.
[image]

Perhaps you didn't properly put the mesh flat on its side. Did you try the Lay Flat feature?

kotoezh, 2016-12-08:
It lays flat, and with 0.1mm layer skirt is ok. But I am using Cura 2.3. Will upgrade now

kotoezh, 2016-12-08:
Please try these steps to reproduce the bug:
1) Set Low Quality, Skirt
2) Load stl
3) Rotate -90 green
4) Lay Flat
5) Rotate -15 blue - here I have a but
Lay flat in this state changes nothing (it is already lays flat)

Ghostkeeper, 2016-12-08:
That reproduces for me in Cura 2.3.0. I'll try again tomorrow in 2.4 when I'm at work. (I don't have the 2.4 development version installed here at home.)

Ghostkeeper, 2016-12-10:
Yup, still reproduces in 2.4.

For some reason, rotating that last 15 degrees messes up the skirt...

kotoezh, 2016-12-11:
Yes, quite a strange behavior. Seems something confuse an algorithm of bound offset construction and makes it go by a wrong line at an intersection. And with Brim where bounds are built from zero distance this does not happen.

BagelOrb, 2017-07-21:
It's most probably not laid flat. Does this still reproduce?

Ghostkeeper, 2017-08-02:
Yes, still reproduces in the 2.7 beta. Same reproduce steps, at least when I try this on an Ultimaker 2.

It's laid flat. It doesn't happen when you don't do that final rotation of -15 degrees. Also, the skirt lines actually intersect with the wall lines:
[image]

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning because it's older than 12 weeks.

BagelOrb, 2017-10-30:
Doesn't seem to reproduce anymore.

[image]

kotoezh, 2018-03-25:
Still reproduces for me with Cura 3.2. On the last screenshot the rotation seems to be by +15deg, and to reproduce you need rotate -15deg:
[image]

BagelOrb, 2018-03-25:
If you could attach your project file then we could try and reproduce this
bug.

Others don't know the settings of your custom fdm printer.

kotoezh, 2018-03-25:
The file is the same that is attached in the first message of the thread, so are the steps to reproduce. The bug does not depend on custom printer settings.

BagelOrb, 2018-03-25:
Reproduced on 3.2.1:
[proto1_mount_socket.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/1845594/proto1_mount_socket.curaproject.3mf.txt)

BagelOrb, 2018-03-25:
@Ghostkeeper  It's a clipper bug in `Polygons::approxConvexHull` O_O

Ghostkeeper, 2018-04-18:
Indeed it seems to be a ClipperLib bug. Here's the skirt before it's inset to the correct distance:
[image]
Notice the spike in the bottom left.

If I simplify the polygon before performing the offset (`simplify(400, 400)`) then the problem goes away. It seems there are microscopic corners in the model.

Ghostkeeper, 2018-04-18:
Fixed here: https://github.com/Ultimaker/CuraEngine/commit/246ed65c3733dc709d10966eeb07020beb02db72

closed 2018-04-18 by Ghostkeeper

kijadesign, 2018-08-09:
the problem surfaced again with version 3.4.1 using a simple 6 sides extruded polygon. using brim would work.
using win 10
part designed with OnShape

[image]

BagelOrb, 2018-08-09:
Please check what the very first layer looks like. You might just need to
pull your object a bit more down.

If that's not the case then please attach the .curaproject.3mf file which
exhibits the problem.

ChrisTerBeke, 2018-08-09:
@BagelOrb might be right. I investigated this issue multiple times before after internal bug reports. Each time it turned out the model was not flat on the bottom...

kijadesign, 2018-08-10:
The part is flat and prints with no issue by just changing skirt to brim, no other change.
Now testing the hints, if I shift it down by the layer thickness, 0.2mm in my case, it fixes the issue.
If I set the layer thickness to 0.3mm and shift only by -0.2mm, the skirt is not generated at all.
weird. there are so many parameters, may be there's some conflict inthere, not all is crystal clear to me.
Notice also that the curvature of the flawed skirt is negative (concave)on 2 sides where the skirt in this case should be all convex, could it be that some variable overflows to a negative value ?

BagelOrb, 2018-08-10:
In order to help you we need to be able to reproduce the situation. Please
save your project as .curaproject.3mf file, sip that file and drag it in a
comment on this issue.

We can then start to help you.

ianpaschal, 2018-08-10:
>  if I shift it down by the layer thickness, 0.2mm in my case, it fixes the issue.

Then I think @ChrisTerBeke and @BagelOrb are correct. And of course, if you use a brim then the brim fills that first layer whereas skirt does not.

BagelOrb, 2018-08-10:
If the skirt is around the wrong areas than the brim should be around the
wrong areas as well.

I would therefore really like to see the project file, to see what's wrong
with the skirt.

kijadesign, 2018-08-10:
here's the file, you can play with layer thickness and offset to investigate.
thks
[polygon_skirt issue.zip](https://github.com/Ultimaker/Cura/files/2278289/polygon_skirt.issue.zip)

BagelOrb, 2018-08-13:
I'm sorry. I didn't look through the history of this issue before posting these last couple of comments.

[image]
This is definitely still a bug in at least Cura 3.4

It is not the kind of bug which magically disappears. We should contact AngusJ about it; perhaps he will fix the bug in the library.

Alternatively we could reimplement our approximateHull function. (It is not the actual convex hull; objects close to each other are surrounded with the same skirt, while objects far away from each other get separate skirts.

Ghostkeeper, 2018-10-11:
The original case (the proto mount, rotated 90 degrees, laid flat, then rotated 15 degrees) doesn't reproduce the bug any more for me in current master. Note that we did update Clipper during the 3.5 development.

Looking at my old screenshot of the bug before performing the inset, if this issue still occurs in different cases, perhaps we can "fix" it by performing the big offset for `approxConvexHull` in two stages by offsetting half the distance each time? If the first offset produces that weird sort of spike in one corner, then the second offset should buff that out.