Ultimaker/Cura #1232: Tower position when center=0,0 (bug?)

karabas, 2016-12-09:
I set center as 0,0
1 Shadow of tower does not know about it at all  and displayed as if center at  0,0 = false
2 Cura accepts only positive values as tower position. So I can place the tower only right far corner of the bed.

Ghostkeeper, 2016-12-09:
Probably the prime locations as well, then.

karabas, 2017-01-11:
The same in 2.4 beta. Tower placement  for delta is very limited due to this bug.
[image]

closed 2017-01-12 by awhiemstra

fieldOfView, 2017-01-12:
As it says in the 2.4 release notes, delta support is largely untested, because none of the Cura regulars have any deltas around. So we rely on reports like this to get things fixed. Thanks @karabas!