Ultimaker/Cura #1233: feature request.

Logo2001, 2016-12-09:
hi Community,

for example:
(Basic Setting: Layer high 0,1mm)
I want to print walls with 0.1mm and all other internal structures in 0.2mm thickness.
This assumes that the wall is printed Twice and the inner once.
I think that would significantly reduce the printing time.

Is this possible or already available?

thanks
Logo

Ghostkeeper, 2016-12-09:
There is the Infill Layer Height setting.

It is not possible to adjust the layer height of any other structures than the infill, at the moment.
And the infill is basically a trick, where it would print infill with for example thrice the line width but only once every 3 layers.

Logo2001, 2016-12-09:
ok, so it is not yet implemented, because i have not found it in the Settings.
"at the Moment" sounds very good :)

thx

closed 2018-02-27 by ianpaschal