Ultimaker/Cura #1238: Cura 2.3.1. can't save profile

johannesritzer, 2016-12-12:
Cura dont let ne save the profile.
[image]

and for sure you will see my log:
[image]

plz help.
cheers!

Ghostkeeper, 2016-12-12:
This happened sometimes when you had custom materials. We fixed that with 2.4. Did you make a custom material, @johannesritzer?

Spirou42, 2016-12-23:
The old profiles for the UM2 for my custom materials are working, but i'm not able to create new ones for my brand new UM3. I have a bunch of custom material settings for nGen, XT, t-glaze and NinjaFlex.

Spirou42, 2016-12-23:
It seams not to be fixed in 2.4b-darwin. Downloaded it and tried to create a profile form the current settings.
I was asked for a profile name and than a couple of files appear in the file system (~/Library/Application Support/Cura/quality) but the user interface is still unaware of the existence of the profiles.
Restarting the application does not help. The profiles are still not visible.

Spirou42, 2017-01-03:
This will not help i presume: In 2.4.0, after adding a couple of more materials (ColorFabb HT and some color variations of nGen) suddenly all my saved profiles appeared.  After this magical appearance all the profiles are working as expected.

RedbullVienna, 2017-03-02:
I'm using facing the same issue with version 2.4.0 :-(
When I create a new material (duplicate and modify) it is still visible in the settings menu (drop down) "material:" but when you click on "manage material" it is gone.
Any fix available yet?

Ghostkeeper, 2017-05-01:
This doesn't reproduce any more, in master at least. If you still see this issue, let us know :)

closed 2017-05-01 by Ghostkeeper