Ultimaker/Cura #1244: Cura 2.X  don't recognize the machine. can't connect via usb

Jakov-Dobie, 2016-12-13:
With lasts 15.x Cura versions the machine is recognized. But in 2.x not, and I can't set the machine options as a communications (Baudrate and serial port).
I'm using a melzi v2 board with marlin, and cura runs on windows 10.
Anyone knows how to fix this?
On 15.x Cura versions I dont have any issues
Thanks

fieldOfView, 2016-12-13:
Since none of the developers at Ultimaker have a printer with a melzi board, they/we are unable to reproduce this, and unable to fix it. 

At this point I would almost consider personally buying a separate melzi board (or some other board that currently does not work) to see what can be done about this, but that's ~60 euro I would rather spend on more fun stuff.

Ghostkeeper, 2016-12-14:
Do you know what baud rate is necessary for the Melzi board? We could add it to the list of baud rates it tries.

fieldOfView, 2016-12-14:
I don't think the baudrate detection is the problem. From posts here and there on the internet it looks like it is the port detection (VCP vs USBSER)

fieldOfView, 2016-12-15:
Found the "posts here and there":
https://www.reddit.com/r/3Dprinting/comments/5g76cs/confused_about_usb_printing_in_cura_231/daq8rk4/

Jakov-Dobie, 2016-12-17:
Thank you all for the help, is really appreciated

bond4u, 2017-02-08:
I have printer running Marlin on Arduino Mega and it was not recognized by Cura. After changing "only_list_usb" to "False" in USBPrinterOutpuDeviceManager.py, it's being recognized now.
But the interface rather useless - I miss pronterface. Want to send commands and move head around.

Ghostkeeper, 2017-02-09:
> I miss pronterface. Want to send commands and move head around.

We had an issue for that! I don't know where it went :(

ianpaschal, 2018-02-26:
This has not been updated in more than a year so I'm considering it `Won't Fix`

closed 2018-02-26 by ianpaschal