Ultimaker/Cura #1245: Unsigned drivers prevent silent install

waoki, 2016-12-13:
Application version: 2.3.1
Platform: Windows 7 Enterprise, 64-bit
Qt: N/A
PyQt: N/A
Display Driver: N//a

Steps to reproduce:

Cura-2.3.1-win64.exe /S

Actual results:

Silent installation stops because the Arduino drivers packaged with Cura are not signed. There's no way to suppress the prompt or to have the installer skip driver installation.

Expected results:

Silent installation completes successfully.

Additional information:

This could be dealt with by a command-line option to skip the "Install Arduino Drivers" section or an updated driver that's signed. The latter is better, but the former would at least allow me to work around the problem with other scripting.

I'm under the impression that the drivers distributed with Arduino 1.6.12 are signed.

Ghostkeeper, 2016-12-14:
I believe we still require Arduino drivers 1.0.x though. @daid do you know this?

daid, 2016-12-14:
Drivers are needed for firmware updates and USB printing, but Cura works fine otherwise without the drivers.

However, I though I drivers where signed these days?

Ghostkeeper, 2016-12-14:
Yeah but we still use drivers from 2013. I think to remember that specifically 1.0.3 was needed for some reason.

thopiekar, 2016-12-15:
I worked on an update maybe one month ago (?), I think @awhiemstra knows the best whether it will go to the next release or not.

ianpaschal, 2018-02-27:
Closing for inactivity (1 year+)

Ghostkeeper, 2018-03-19:
This was fixed for Cura 3.2. We've updated the Arduino drivers which they now sign properly.

davleema, 2018-04-26:
Update:
Tried today with latest release (3.3.1) : same problem, seems to show even more popups (5 to 6).
The most recent version that installs without popups is 3.1.0.

thopiekar, 2018-04-26:
Hmm, once the drivers are signed, they can also get out-of-state? Like SSL
certificates?

Never noticed that, but can imagine that is the case.

Ulyssys, 2018-05-08:
Also here to voice my concern as this is a problem for automation. Hopefully a permanent fix can be made to the installer.

Ghostkeeper, 2018-05-11:
We should consider using CMake to download and include the Arduino drivers into the installer, instead of putting it cura-binary-data manually. It'll be much easier and safer to update those drivers then.

davleema, 2018-06-07:
Any news yet on this issue ?

ianpaschal, 2018-07-02:
Not really. Since the only effect is that you have to press "OK" sometimes this issue gets significantly lower priority than, say, crashes or corrupted files, etc.

Since the older tickets have been archived when we thought it was fixed, there's a new ticket for this internally: CURA-5079

davleema, 2018-07-02:
Well, it may not be a huge issue to you, but it does prevent us from running a completely unattended deployment of the latest version of Cura. So, if you could fix this, we would really appreciate the effort.

ianpaschal, 2018-07-02:
We will, we actually just discussed it today, but please understand that it's a relatively uncommon use case. And every bug is important to someone. 😉

damon-atkins, 2018-07-22:
Tried Silent Install (/S) of 3.4.1 which should be no GUI had the following appear

- First Prompt

> [image]

- Second Prompt
>[image]

Ghostkeeper, 2018-08-17:
From those screenshots it looks like they are signed fine and the certificate is still okay.

I tried looking for a solution and came upon this: http://www.migee.com/2010/09/24/solution-for-unattendedsilent-installs-and-would-you-like-to-install-this-device-software/
Maybe a solution for us would be to register that certificate before trying to install the driver. That is quite a big security hole in Windows if it works though. It defeats the entire purpose of the pop-up.

davleema, 2018-08-22:
Yep, this workaorund seems to solve the problem of the pesky popups. Importing the certificates prior to launching the Cura installer avoids the popus from appearing, and thus gives us a completely silent installation. 
My installation script now looks like this:
```
 REM -- install certificates to get rid of popups for Adafruit/Arduino drivers 
 certutil -addstore TrustedPublisher "%~dp0adafruit.cer"
 certutil -addstore TrustedPublisher "%~dp0arduino1.cer"
 certutil -addstore TrustedPublisher "%~dp0arduino2.cer"
 timeout /T 2
 
 REM -- Run Cura 3.x silent installation
 "%~dp0Cura-3.4.1-win64.exe" /S
 timeout  /T 3
```
This workaround shouldn't be necessary, but hey, it works for now.

jmerriweather, 2018-09-02:
Is this simply a case where the IfSilent is not working in installer.nsi?
Should 
```
  ${If} ${RunningX64}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst64.exe" /lm'
  ${Else}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst32.exe" /lm'
  ${EndIf}
```
be
```
  ${If} ${RunningX64}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst64.exe" /lm'

  ${Else}
    IfSilent +2
      ExecWait '"$INSTDIR\drivers\dpinst32.exe" /lm'

  ${EndIf}
```

thopiekar, 2018-09-03:
Hmm, I don't see a direct difference between both code blocks, except that you added empty lines after ExecWait. Do you know whether it makes a difference?

jmerriweather, 2018-09-03:
@thopiekar I just ran a simple test and the spacing isn't the issue:
```nsis
# set the name of the installer
Outfile "hello world.exe"

!include "MUI2.nsh"
!include "Library.nsh"

# create a default section.
Section

${If} ${RunningX64}
  IfSilent +2
    MessageBox MB_OK "NOT Silent"
${Else}
  IfSilent +2
    MessageBox MB_OK "NOT Silent"
${EndIf}

MessageBox MB_OK "Silent"

SectionEnd
```
The "NOT Silent" isn't displayed when /S is used. Not sure why ExecWait still executes even when the silent switch is set.

ZaxLofful, 2018-10-11:
I am attempting to install CURA with SCCM 2016 and I am still running into this problem. I amusing the Cura 3.5 64-bit installer.

It sounds like a solution has been found, when is this going to make it to release?

basictheprogram, 2018-10-25:
I'm a co-maintainer of the chocolatey package and I'm still running into this problem as well. 
On the chocolatey-test-environment server, Windows Server 2012 R2 Datacenter Build 9600, testing things manually results in the Window Security popups.

```shell
.\Ultimaker.Cura-3.5.1-win64.exe /S
```
[image]

[image]

[image]

thopiekar, 2018-10-25:
I don't think you can actually prevent these. I think it would be easier, if we find a way to skip installing these, when we are in silent mode. Think that is easier for now, I think.

fieldOfView, 2018-10-25:
That installer.nsi is not used, nor has it been for a fairly long time.

basictheprogram, 2018-10-25:
I think this will ignore the Arduino Driver installation, that is 9e7951d3a88b41f8f0421e2832f4fd6a34a7b6f7 does. Sorry, no way to test it. 
```
  IntCmp $arduino_selected 0 +6
  ${If} ${RunningX64}
    IfSilent +2
      ExecWait '"$INSTDIR\arduino\dpinst-amd64.exe" /lm /sw'
  ${Else}
    IfSilent +2
      ExecWait '"$INSTDIR\arduino\dpinst-x86.exe" /lm /sw'
  ${EndIf}
```

thopiekar, 2018-10-25:
@fieldOfView Thats correct. The nsi is generated by CMake inside the cura-build project.

ZaxLofful, 2018-11-16:
Can we push this through? I need this badly for my SCCM deployment!

basictheprogram, 2018-11-29:
3.6.0 still does not do a silent installation.

https://github.com/Ultimaker/Cura/releases/download/3.6.0/Ultimaker.Cura-3.6.0-win64.exe

[image]

Looks like the support for skipping the arduino stuff is in the template

https://github.com/Ultimaker/cura-build/blob/master/cmake/NSIS.template.in

I'm guessing 3.6.0 did not get built with https://github.com/Ultimaker/cura-build/pull/199/commits?

Appesteijn, 2018-11-29:
@basictheprogram Your guess is right, the 'silent install' was done after 3.6.0 was released.

ZaxLofful, 2018-11-29:
Can we get an ETA on what release this will make it into?

I know y’all are busy with other features, but allowing this would get your product into so many more people’s hands....At the very least 150 more people in my company are looking for this and I don’t want to install it on each machine by hand.

Appesteijn, 2018-11-29:
We plan for a next beta end of this year/starting next. If you can build Cura yourself, you can do it already.

MochaHero, 2018-12-05:
I got a copy of the arduino folder and exported the certificates into a folder to be accessed by my script before installing the drivers and then running the Cura Installation. 

Not sure if it's really the best way, but it worked for me.

```

#Script Locations
$ScriptPath = $MyInvocation.MyCommand.Path
$ScriptDir = Split-Path $ScriptPath

#Certificate Locations
$certCheckLocation = "Cert:\LocalMachine\TrustedPublisher\(Cert Thumbprint...)"
$certPutLocation = "Cert:\LocalMachine\TrustedPublisher"
$certFile = (Get-ChildItem -Path "$ScriptDir\Arduino Certificates\*" -Filter *.cer)
$certFilecheck = (Test-Path -Path "$ScriptDir\Arduino Certificates\*" -Include *.cer)
$certCheckStore = (Test-Path -Path $certCheckLocation)

#### Import Driver Certs for Cura Software ####

if ($certFilecheck -eq $true) {

     Foreach ($cert in $certFile){
        try{
            $cert | Import-Certificate -CertStoreLocation $certPutLocation
            Write-host "$cert successfully imported."
        }
        catch {
            Write-Error ("Error adding '$certfile' to '$certPutLocation': $error") -ErrorAction:Continue
                $error.Clear()
        }
    }
    
}

```

Then my script will run "dpinst-amd64.exe" to install the drivers silently

```

#### Begin Arduino Driver Install ####
Write-host "Begin Arduino Driver Install..."

    try {
        Start-Process "$ScriptDir\arduino\dpinst-amd64.exe" -ArgumentList "/S /F" -Verbose -Wait
    }
    catch {
         Write-Error ("[ERROR]`t Error installing Arduino Drivers. $error") -ErrorAction:Continue
            $error.Clear()

```

opsi-archIT-KIT, 2019-01-30:
> We will, we actually just discussed it today, but please understand that it's a relatively uncommon use case. And every bug is important to someone.

I don't agree, that silent (de)installation is an uncommon use case. Every company or institution with professional structures will use a software deployment system like SCCM, chocolatey, OPSI or many others. All these need this 'silent' option. The NSIS installer therefore provides a silent option per default.

Actually I wanted to create an OPSI package for cura 3.6.0 for our institution. I can confirm the popups on installation. I want to enlarge this issue that silent deinstallation also is not working!

[image]

thopiekar, 2019-01-30:
Agree, and since multiple Cura versions can be installed, it need a way to clean up the older versions. Otherwise you end up with a new installation per approx 2 months.
Guess that there are many strict companies, where employees normally don't have admin rights, so they could clean it up on their own.

opsi-archIT-KIT, 2019-01-30:
@thopiekar: deinstallation of older version could be feature of the installer, but it is not a must.

Most software installer packages do not care about old versions, the deployment system should/can take care of this

Ghostkeeper, 2019-02-01:
Jira shows that there was still activity on CURA-5927 last week. It's still in progress and not decayed yet. We're on it.

tkhumphreys, 2019-03-11:
Certificates Arduino LLC and Arduino srl are currently out of date, Arduino LLC expired 19/01/17 and Arduino srl on 06/01/19. When the certs are installed prior to running the installer, the installer ignores the fact that these certificates are pre-installed and prompts for a "Always Trust this software" dialog box during a silent install or a manual install. Are there current versions of these certificates available?

LipuFei, 2019-08-20:
The signatures of the Arduino drivers are ok. Although the certificates of Arduino LLC and Arduino srl expired in 17, the signatures were made when the certificates were valid/active, so the signatures themselves are still valid. The problem is that those certificates are by default NOT in the trusted publishers list in the certificate store of Windows 10. So, the first time you try to install a driver signed by that party, the confirmation dialog will pop up. There are 2 ways to fix this:

 - Make the arduino drivers optional so you can skip them when you want to install Cura silently.
 - Import the arduino certificates as trusted publishers before you install Cura. To do this, you can use the [Import-Certificate](https://docs.microsoft.com/en-us/powershell/module/pkiclient/import-certificate?view=win10-ps) cmdlet from PowerShell with `Import-Certificate -FilePath "my-cert-path"`. You will need 3 certificates for this:
    - Adafruit Industries
    - Arduino LLC
    - Arduino srl

[image]

Ghostkeeper, 2019-08-21:
Adding trusted certificates is quite a common task for an IT department so maybe that is okay as a workaround then?

LipuFei, 2019-08-21:
This issue is actually fixed in https://github.com/Ultimaker/cura-build:

https://github.com/Ultimaker/cura-build/blob/ec72bcd9c6e8e39c4e23e508613922c3d28cef3f/cmake/NSIS.template.in#L761-L767

I just tested with a newly installed windows server 2019. Downloaded Cura 4.2.1, and installed it by running `Cura-4.2.1-installer.exe /S`. It performed a silent installation successfully without installing the Arduino drivers, thus no pop ups. After that, when I tried to manually install the Arduino drivers, I still got the confirmation pop ups.

I will close this ticket then.

closed 2019-08-21 by LipuFei

craigkittelsonNWTC, 2021-11-16:
Hello, I am having this same issue with the installer for Cura version 4.12.0 x64 on Windows 10 64-bit. Silent install creates four prompts to install drivers.

Ghostkeeper, 2021-11-24:
@craigkittelsonNWTC Please see a more recent report of this issue: https://github.com/Ultimaker/Cura/issues/10091