Ultimaker/Cura #1247: AttributeError: 'CuraContainerRegistry' object has no attribute 'lockFile'

peteruithoven, 2016-12-14:
I was trying the cura-master ppa from thopiekar, which I totally understand is unstable, to check whether https://github.com/Ultimaker/Cura/issues/1223 is fixed. 

When booting it gives a fatal exception, with a very cute kitten and the following information: 
```
Version: 2.3.99-master
Platform: Linux-4.4.0-53-generic-x86_64-with-elementary-0.4-loki
Qt: 5.5.1
PyQt: 5.6

Exception:
Traceback (most recent call last):
  File "/usr/bin/cura", line 61, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 310, in getInstance
    Application._instance = cls()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 208, in __init__
    with ContainerRegistry.getInstance().lockFile():
AttributeError: 'CuraContainerRegistry' object has no attribute 'lockFile'
```

Ghostkeeper, 2016-12-15:
I think the Uranium version the PPA uses may be outdated then.

cura.Settings.CuraContainerRegistry inherits from UM.Settings.ContainerRegistry, which has the lockFile function: https://github.com/Ultimaker/Uranium/blob/master/UM/Settings/ContainerRegistry.py#L507

@thopiekar Do you know whether the Uranium is in sync with Cura on your PPA?

thopiekar, 2016-12-15:
Hmm, we use the lockfiles for weeks. The only delay can be max. one day, since launchpad is building packages daily.

I checked the website and it even says:
[image]

Here is the list of packages for 16.04 aka Xenial: https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master/+packages?field.name_filter=&field.status_filter=published&field.series_filter=xenial

What you should probably check is whether there are no blocked upgrades or mixed up packages with stable.
2 or 3 weeks before I made mistake and packages from stable are now "newer" to APT then the one in master. Maybe you'll need to remove all packages from the PPA first and reinstall. Alternatively you can also use APT pinning, but I would only prefer it if you are already familiar with it.
Hint: A helpful tool to manage packages is Synaptic. I already used this application from the beginning I used Ubuntu (which was the first distro I used).
[image]

thopiekar, 2016-12-15:
Oh, forget about the versioning problem. 2.4 has just been released..

Time to update the stable PPA :dancer:

thopiekar, 2016-12-15:
@peteruithoven Btw. your fix should be in 2.4, isn't it, @Ghostkeeper ?

peteruithoven, 2016-12-15:
Awesome! I'll try with the new stable. 
Next time I'll try to remove all packages from a ppa before switching to the next one.

closed 2016-12-15 by peteruithoven

thopiekar, 2016-12-15:
No problem. I started to make a summary about troubleshooting here:
https://github.com/thopiekar/Cura-packaging/blob/master/Troubleshooting.md

If you have other helpful tricks, which might help other users, feel free to edit the file and send a PR.

(Updating the stable PPA might take some time. Better then doing it to fast and making mistakes again :smile: )

Regards!

Ghostkeeper, 2016-12-15:
We created the branches for the 2.4 release this afternoon. I think that everything got into those branches properly. At least Github says that the branch is 5 commits ahead of Master, and no commits behind.

But I think that we forgot to merge 2.4 into master as well. Multiple of us. We aren't used to the workflow any more since that 2.3 release ;)

thopiekar, 2016-12-16:
Good news! Yesterday I finished to update the build recipes. I only have to wait for feedback on 1 or 2 plugin projects, whether they want to create a stable ("2.4") branch or not.