Ultimaker/Cura #1248: No options for UM Polycarb material?

partsoven, 2016-12-15:
I bought some PC listed on Ultimaker's website, however the UM firmware doesn't have an option to load it, and there are no profiles for it in Cura.

Ghostkeeper, 2016-12-15:
PolyCarbonate is only supported for the Ultimaker 2+ at the moment. We haven't made any profiles for it yet in Cura for any other printer of ours.

I suggest that you select Nylon in the printer, which has the most similar settings to PC for the things that the firmware does (heating nozzles for material switches, that sort of thing). Then in Cura, you can try to create a profile that mimics the important settings of the PC profile for UM2+.

Additionally, the UM2+ firmware uses a bed temperature of 110 degrees C and a printing temperature of 260 for PC materials. That won't be listed in the Cura profiles because Cura can't control the printing temperatures for UM2+. The source for the UM2+'s PolyCarbonate settings is here: https://github.com/Ultimaker/UM2.1-Firmware/blob/aefc22adb21246c729ec0b8d9833a330757068c4/Marlin/UltiLCD2_menu_material.cpp#L1103

Ghostkeeper, 2016-12-15:
I'm sure @THeijmans can give you tips on how to print properly with PC on an Ultimaker 3. He's good at that sort of thing and is one of the people that creates our profiles.

fieldOfView, 2016-12-16:
To clarify: it is not *making* the profiles that takes much work (it is mostly just editing values in Cura), it is testing them and making sure that they are the very best you can get out of the printer with that material that takes time.

partsoven, 2016-12-16:
Ok. I will just wait until profiles become available and the firmware is updated to use PC.

closed 2018-02-27 by ianpaschal