Ultimaker/Cura #1249: Cura GUI not working on Linux

XenGi, 2016-12-15:
Application Version: 2.3.1
Platform: Linux
Qt: 5.7.0
PyQt: 5.7
Display Driver: Intel 2.99.917+746+g169c74f

Steps to Reproduce:
  - Install cura
  - Start Application

Actual Results:
  - Add printer wizard launch on every start,
  - non working GUI elements like dropdowns and menus
  - 3D Print area not working

Expected results:
  - add printer only once
  - working gui elements
  - working 3d area with print

Additional Information:
[image]

Ghostkeeper, 2016-12-15:
For some reason it seems to fail to add a printer or fail halfway through. As a consequence, a lot of GUI elements are empty or broken.

Could you show us your log file? It should be in `~/.local/share/cura/stderr.log`. That will help us debug your problem.

XenGi, 2017-01-02:
Here's the cli output when I start the program after a fresh install and choose a Ultimaker 2+ as my printer: https://gist.github.com/XenGi/51c407a0ebf81ffbd266e28f3960937c

The file `~/.local/share/cura/stderr.log` doesn't exist. instead I have a `~/.local/share/cura/cura.log` which matches the output on the terminal.

Ghostkeeper, 2017-01-02:
> 2017-01-02 11:46:46,846 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]): AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'
> fish: “cura” terminated by signal SIGSEGV (Address boundary error)

:fearful:

XenGi, 2017-01-07:
I tried the AppImage version of Cura 2.4 Beta and it worked. So maybe in future version this bug is not important anymore.
But still it would be nice to find out why this doesn't work and hopefully find a way to fix it.
Do you have any clue what could cause this?

awhiemstra, 2017-01-09:
The logs suggest it does not have a proper machine definition, which then cause a bunch of cascading errors. I am not sure /why/ it does not have a proper machine definition. One thing I can think of is that we cache the binary representation of those files and your cache got corrupted somehow. This would explain why it works with 2.4 as it uses a different cache directory.

XenGi, 2017-01-09:
I thought it works with 2.4 because it's an AppImage and brings all it's libraries and little rootfs. I don't know how AppImages work exaclty but I assume they are completely independent from my host system.

awhiemstra, 2017-01-09:
That is another option. It is a bit tricky to verify what is happening so I think it is mostly academic at this point.

Note that the AppImage does include a bunch of libraries but not everything. We assume a certain base system (CentOS 6 in our case) and everything included in that is not included in the AppImage.

closed 2017-07-27 by nallath