Ultimaker/Cura #1250: Extra Fill-In Layers in 3D Print ???

SolidWorksMagi, 2016-12-16:
Hi, 

I'm using CURA 2.3.0 and the hollow rhombicosidodecahedrons I'm printing have extra "fill-in" layers
http://brainless.org/MultiMedia/Pictures/IMG_6014-20161215-RhombiCosiDoDecaHedron-CURA%20Print%20Faults-ec800.JPG

The upper and lower thin plates are not in the original design/.STL file I created;
http://brainless.org/MultiMedia/Pictures/RhombiCosiDoDecaHedronT.jpg

Ghostkeeper, 2016-12-16:
If you use X-Ray View in Cura, and you move the camera around a bit, does anything become red?

If so, you have extra faces in your model or missing faces, and it becomes impossible for Cura to reliably detect what is the inside of your model and what is the outside. That is often the cause of this kind of filling.

SolidWorksMagi, 2016-12-16:
HI, 

 

I've printed these same parts on other 3D printers using their software, 3D Touch, and Cube2 and do not get those two "plates" inside the figures.

 

 

The original design is in SolidWorks and does not have any hidden plates/planes etc… I created a 3D Structure exactly like this.  

 

 

You can see on the bottom I've added a 10mm x 0.25mm "raft" so that the print better stays in place while printing to reduce errors.  That's the only flat surface wider than any cross member in the design.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

BerndJM, 2016-12-16:
Just an idea:
what are your settings in "Mesh Fixes"?

fieldOfView, 2016-12-16:
I have a model that consistently reproduces the issue. Turning off "Union Overlapping Volumes" (which is on by default) fixes the issue.

SolidWorksMagi, 2016-12-16:
Hi, 

 

Okay, can you give me a full list of clicks to find that setting?  "Union Overlapping Volumes"?  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

SolidWorksMagi, 2016-12-16:
Hi, 

 

How do I find;  "Mesh Fixes" settings?  All my settings are in the attached image.   

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

fieldOfView, 2016-12-16:
Image attachments don't work through email on github.

Move over one of the setting categories, and press the cogwheel that appears
Either scroll to find the "Mesh Fixes" category, or:
In the "filter" textbox, type "Union"
See the "Union Overlapping Volumes" setting appear before your eyes.

BerndJM, 2016-12-16:
Attached image ?!?
"Union Overlapping Volumes" is a setting in "Mesh Fixes".
If it's not visible go to:
Preferences -> Configure Cura -> Settings (Visibility)
Scroll down if you find "Mesh Fixes" and enable the settings in this category
after closing the dialog they will be visible in the settings on the right side.

SolidWorksMagi, 2016-12-16:
Hi, 

 

Mesh Fixes = UnChecked.

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

fieldOfView, 2016-12-16:
"Mesh Fixes" is not something that can be checked. "Union Overlapping Volumes" is. We're telling you to check it in the Setting Visibility pane. The sidebar will then get a new category "Mesh Fixes" which has the setting "Union Overlapping Volumes" in it. That setting *in the sidebar* will be checked, because it is checked by default, if it is visible. Please uncheck it *in the sidebar*, and see if that fixes your problem.

SolidWorksMagi, 2016-12-16:
Hi, 

 

Okay, under Preferences - Configure CURA - Settings - Mesh Fixes … I added a check to the Union Overlapping Volumes …

 

That added a new Mesh Fixes in the Print Setup Advanced List on the right side with the " Union Overlapping Volumes" being checked.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

Ghostkeeper, 2016-12-17:
You've now made the "Union Overlapping Volumes" setting visible. As fieldOfView said, it is enabled by default. Disabling this setting may be a workaround to your problem, so uncheck the box that just appeared.

SolidWorksMagi, 2016-12-17:
Hi, 

 

Wasn't it already unchecked before I added the check mark in the Preferences - Configure CURA - Settings - Mesh Fixes - Union Overlapping Volumes?

 

That's when I was getting the extra thin layers.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

 <http://www.R2Pv1.com/> http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

BerndJM, 2016-12-17:
With the checkmark in "Preferences ... Settings **Visibility**" you control the **visibility** (in the right bar) of the different settings.
And this setting "Union Overlapping Volumes" is (as all the others said before!) enabled (checked) by default. So I repeat what Ghostkeeper said:

> Disabling this setting may be a workaround to your problem, so uncheck the box that just appeared.

SolidWorksMagi, 2016-12-17:
Hi, 

 

I did UnCheck it and will do a test print tomorrow after a current print job is completed.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

SolidWorksMagi, 2016-12-17:
Hi,

 

New print with "Union Overlapping Volumes" UnChecked still has the two extra planes.

 

I have the three files on:  https://pinshape.com/items/30014-3d-printed-rhombicosidodecahedron-snowpersons 

 

 

Here you can see the correctly printed snowballs from other 3D printer GCode generators;  

 

 

 

 

Here you can see the screwed-up printed snowballs from CURA 2.3.0;  

 

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

 <http://www.R2Pv1.com/> http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

BagelOrb, 2017-07-23:
You should check the layer view rather than actually doing a print.

Instead of fiddling with the mesh fixes you should go back to your modeling program and fix the problems in your model. Check the X-ray view in Cura to see if the problem is still there.

closed 2017-07-23 by BagelOrb