Ultimaker/Cura #1252: Camera control bugs (Cura 2.3.1)

Ando120, 2016-12-17:
It looks right place to report Cura bugs
And sorry about my bad English.

I found 2 bugs with camera control and made fixes in "CameraTool.py" file

### 1. Zoom in bug. 

When you continue zoom in after you don't see visual difference, then you must zoom out same amount after you will see visual zoom out.

Fix: In CameraTool.py "self._min_zoom = 0" but mathematically it will move endlessly closer to 0. 
So min zoom should be example 1
"self._min_zoom = 1" is fixing that bug

### 2. space + mouse zoom jump
Space + mouse zoom "jump"(zoom in or out) sometimes when start new zoom. ( only with "space + mouse" zoom) 

Fix: Script don't reset sometimes "self._start_y" value before next zoom.
added reset when "space" is not active. 

```
    def initiateZoom(self, event):
        if event.type is event.MousePressEvent:
            return False

        #added 2 lines to fix that bug
        elif event.type is Event.MouseMoveEvent and self._space_is_active is False: #space -> mousemove
            self._start_y = None

        elif event.type is Event.MouseMoveEvent and self._space_is_active is True: #space -> mousemove
```

Ghostkeeper, 2016-12-17:
> It looks right place to report Cura bugs

It is! Thank you. I'll add this to our tracker so that it gets planned.

You seem to have already made some steps to try and fix the code. You could make those changes with a pull request, if you like. We'll then review them and test them and you'll get proper attribution, your username being forever lodged in our commit history. Otherwise we'll eventually make these changes ourselves.

nickthetait, 2016-12-19:
Here is instructions on making a pull request https://help.github.com/articles/creating-a-pull-request/

Ando120, 2016-12-21:
Still don't understand how to "add commits" ?  There isnt file plugins\Tools\CameraTool\CameraTool.py

Ghostkeeper, 2017-05-01:
This issue was closed because it had been inactive for too long.

Ghostkeeper, 2017-08-10:
The bug still exists.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
The spacebar zooming issue seems to have been fixed now.