Ultimaker/Cura #1253: Importing a material compatible with "fdmprinter" results in the deletion of all printer profiles.

gvJaime, 2016-12-17:
When you manually configure a material with a <machine> compatibility instance with "product=fdmprinter" in it, cura will delete (or simply not display) all printer profiles.

A small peek in the code made me think that all printer definitions inherit from fdmprinter, so a material compatible with fdmprinter would be compatible with any printer.

I have [this material file](https://github.com/gvJaime/Cura-Settings/blob/master/BlackBQPLA.xml.fdm_material) from one of my repos that illustrates the issue. (Warning. It DOES delete all the printers after you add it)

nallath, 2016-12-19:
FDMPrinter is an "abstract" printer. It doesn't actually exist. We never accounted for fdmprinter being instantiated, so i'm not that suprised you get all kinds of weird results.

It is a bit nasty that it suddenly deletes all your old data though. Could you also post your log files?

gvJaime, 2016-12-19:
OK, I have this [before.log](https://gist.github.com/gvJaime/8887f22ed3ee3c7db0ee5effad7fc1bd/raw/53bdbeb4df919d23d6b7b2a90fe0f90983e81de9/before.log), in which I open Cura with a ulti3 and a bq hephestos XL configured, and I import a material compatible with the fdmprinter only. I activate it (because it appears on the list), and close the program.

And then I have this [after.log](https://gist.github.com/gvJaime/f99c90eaa51647c036d79c5bf1456cff/raw/446669e15030d049ab66e576b269394ce2dd0b99/after.log) in which I open Cura again, and I find that I have to configure the printers over again, because they are gone. Configure an Hephestos XL, and close the program.

In the after.log there is an entry that says "Could not deserialize BQ Prusa Hephestos i3 XL", followed by a python traceback. Maybe that's the cause.

I'm not sure these are the logs you asked me for, I had to copy the terminal output by hand because for some reason the command `cura >> somename.log` didn't work.

Thank you

nallath, 2016-12-19:
It's not the only thing that is wrong with your Cura. For some reason it seems to be unable to find the network printing plugin

```
2016-12-19 12:09:52,411 - ERROR - .../UM/PluginRegistry.py (_findPlugin [268]): Import error loading module UM3NetworkPrinting: cannot import name 'NetworkPrinterOutputDevice'
2016-12-19 12:09:52,411 - ERROR - .../UM/PluginRegistry.py (_populateMetaData [238]): Could not find plugin UM3NetworkPrinting
2016-12-19 12:09:52,413 - ERROR - .../UM/Qt/Bindings/PluginsModel.py (_update [34]): Plugin is missing a plugin metadata entry

```
It also mentiones that there is a model not found;
```
file:///opt/cura/share/cura/resources/qml/AddMachineDialog.qml:105: ReferenceError: sortedMachineDefinitionsModel is not defined
```
I don't know why it's doing that, but the current code has no reference to a sortedMachineDefinitionsModel (so that should already resolve some issues you might be having)

fieldOfView, 2016-12-19:
The sortedMachineDefinitions thing was fixed recently with this PR:
https://github.com/Ultimaker/Cura/pull/1241

gvJaime, 2016-12-19:
I'm using 2.3.1-2. So maybe in the next release this will be fixed.

I'm using "fdmprinter" in the machine compatibility entries because none of the names of the Hephestos XL worked for the `product=` field. I tried with either the name (BQ Prusa i3 Hephestos XL) or the id (bq_hephestos_xl)

Nothing. It doesn't recognize any of those, and I don't know where to look for the proper machine identificator.

Is there anywhere I can look for the proper identificator, so I can make the material files compatible with my machines? I already checked the .json file, thats where I found the id = "bq_hephestos_xl"

Ghostkeeper, 2016-12-20:
It's doing some [strange things](https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L471) to the machine identifier before looking it up in the list of definitions. But I'd expect `bq_hephestos_xl` to work...

nallath, 2016-12-20:
Directly using the ID in the product name should work. I've tried that a few times.

fieldOfView, 2016-12-20:
Note that you should not have to add a section for the bq_hephestos_xl, unless you changed the bq_hephestos_xl definition. The Ultimaker 2(Extended)+  and Ultimaker 3 (Extended) have a special metadata entry "has_machine_materials", which most other printers don't have. As far as I know the code (but it may have changed), only definitions that have a "has_machine_materials" set to true need a section in the material xml file.

gvJaime, 2017-01-08:
@fieldOfView , by bq_hephestos_xl definition you mean [this definition](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/bq_hephestos_xl.def.json)? If so, I'm afraid that "has_machine_materials" is not set to anything, I'm assuming that it defaults to false.

When you make a custom RepRap config, I'm assuming that the new printer doesn't have that field set to true, but it still doesn't show up.

Has there been any changes regarding this issue?

arthurlutz, 2017-01-09:
I think I'm bumping into this problem too. 

Am trying to use the community configuration for Dagoma's DiscoEasy200 from https://github.com/DelphinPETER/Dagoma-Settings-For-Cura

I get (even when I add "has_materials" or "has_machine_materials")

[image]

alekseisasin, 2018-05-08:
Hi @gvJaime , I think the issue is solved in the latest Cura 3.3.1 version and you can make/add simply new printers. If you still want to add a new printer you always can make a Pull Request which will be checked.

gvJaime, 2018-05-22:
Hello @alekseisasin , I thought this was closed already! It haven't done any more research on this, but I had to reimport all my settings a couple of times in the last four months and had zero problems.

I think it is safe to close this issue. I can do it for you.

Thank you!

closed 2018-05-22 by gvJaime