Ultimaker/Cura #1254: Cura 3.2.1 does not save printer settings

whoisnosferatu, 2016-12-17:
I have gone to the new version 3.2.1 and when I save the configuration for my prusa i3 steel and the settings of the machine when closing the program and to reopen it again it asks to me what type of machine I have, it does not contain the configurations of the type of printer.
Always when you reopen the program you are prompted for all the data on the printer, besides not storing the nodifications in starcode lines like g29 when closing and opening the program are not saved setings

Ghostkeeper, 2016-12-18:
I think he meant 2.3.1.

Ghostkeeper, 2016-12-18:
I can't reproduce it though. I tried adding all of the Prusa printers and closing and re-opening in between, but it properly saves all settings, and doesn't ask me to add a new printer.

Did you make any other modifications after you added the printer?

fieldOfView, 2016-12-18:
@whoisnosferatu, do you give your printer a name? If so, is there a comma or other diacritical in that name? Or do you create a profile with a comma or other diacritical in its name?

whoisnosferatu, 2016-12-18:
Sorry, I was referring to version 2.3.1
I finally uninstalled and installed 4 times and it seems that now saves the profiles and does not ask me for the type of printer to start the program now test, it seems that also save the configuration of the start gcode now will print a piece if I detect the g29 for Do a z probe.

I had to uninstall also all previous versions of cure that had version 15.xxx and another 20.xxx that had installed together in the computer
Thank you, I will continue to try and inform you

mikefazz, 2016-12-23:
I had the same issue... after lots of messing around it seems the ID and Name of the printer need to be different... 

I changed this in the ..resources\definitions folder

Ghostkeeper, 2016-12-27:
What did you change to fix it? Do you remember?

mikefazz, 2016-12-27:
If you find the name and ID of your printer in the text file for your printer they need to be different.  But there could be some other issue.  I think that if the *.def.json file for your printer doesn't follow curas expected style it basically doesn't close properly... having readable text files to allow adjusting settings is nice.  Being able to change ALL the settings within the GUI would be better.  Setting up a custom delta with 3 extruders took a while to figure out...

look in:
C:\Program Files\Cura 2.3\resources\definitions

ret1, 2017-01-14:
ive got the same problem when i use commas in profile names. dont use commas at all (in names) because its also the field seperator for items, saved in the configuration files.

awhiemstra, 2017-01-16:
The comma in profile names was fixed for 2.4. It should be unrelated to commas in the machine name.

closed 2017-07-27 by nallath