Ultimaker/Cura #1255: Cura 15.04.6 (possibly onward?) Incorrect Bridging Tool Path/ GCode

TpForMyBungholio, 2016-12-18:
I've been attempting to do some 50 hour prints, and I've been tearing my hair out trying to figure out why layers that bridge seemed to suddenly get wider. Well, come to find out that that's exactly what Cura was telling them to do. I've attached a screenshot of the problem. If there is a bridge layer, Cura instructs the toolpath the move out beyond the edge of the part. I'd ballpark it at maybe 1mm error? Error exists uniformaly on all sides of the part.

This error has cost me a lot of time and money, and I'd appreciate it if it could be addressed.
[image]

sedwards2009, 2016-12-18:
Please try again using a recent version of Cura.

thopiekar, 2016-12-18:
@TpForMyBungholio , sorry to hear about your problems with Cura.

In case you have to use strictly Cura 15.x: 
Issue reports of Cura version like 15.x and older should be reported at: https://github.com/daid/LegacyCura

Regards

daid, 2016-12-19:
Also, there is no bug. What you see is a rendering "artifact" the top X layers of the viewer are rendered in full, the rest is rendered as thin lines to speed up the rendering.

closed 2016-12-19 by daid