Ultimaker/Cura #1256: Low Camera FPS

SQuicke, 2016-12-19:
When monitoring my ultimaker 3 prints on the LAN the camera FPS is down to around 1-2fps.  This is using my Laptop using a wired connection.  I've recently downloaded the ultimaker 3 app and monitoring the camera on my phone over wifi is much smoother.  Is there any known issue with the camera or could it be a problem with the LAN?  Everything else seems to run fine over the network.

daid, 2016-12-19:
Known issue, there where problems in getting the same feed working in Cura as the App is using. So it falls back to using slow snapshots. I think the next update will fix this problem. (I'm sure someone will drop in here to confirm this)

SQuicke, 2016-12-19:
Thank you David.  Any ideas when the update will be released?  It's not a major issue, more frustrating than anything!

daid, 2016-12-19:
I hope before new year.

fieldOfView, 2016-12-19:
Cura 2.4 does indeed use the video stream instead of snapshots. It also stops requesting frames when you're not looking at the monitor (Cura 2.3 is a bit wastefull there)

daid, 2016-12-19:
How does Cura know I'm not looking at the monitor? Cura spying on me?!? ;-)

nallath, 2016-12-19:
Yes

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-10:
Fixed in firmware.

fieldOfView, 2017-08-10:
No, fixed in Cura