Ultimaker/Cura #1257: 2.4 - Project saved with no file extension

smartavionics, 2016-12-19:
Just trying out 2.4 on a 64 bit Ubuntu system and when I saved a project it didn't give the file the .curaproject.3mf extension. I would have thought it should have done as the file save dialog has that file type filter.

nallath, 2016-12-19:
This sometimes happens. I've not been able to figure out where this comes from. I get the impression that this has to do with the file dialog in Ubuntu.

BerndJM, 2016-12-21:
Additional:
after saving a project (without extension) the regular "Save to File"  switches from .gcode to .3mf as default ending.
When you now again save the project, the filename is correctly appended with .curaproject.3mf
...
spooky ;-))

Ghostkeeper, 2016-12-22:
Was that also on Ubuntu, @BerndJM? Or on OSX? Because on OSX we save simply as .3mf due to OS constraints.

BerndJM, 2016-12-22:
It's on Ubuntu ...

ChrisTerBeke, 2017-11-14:
Is this still an issue?

smartavionics, 2017-11-14:
Don't know, let's close this issue and if I have any more problems I will create a new issue.

closed 2017-11-14 by smartavionics

Ghostkeeper, 2017-11-24:
We had some fixes for this around the release of Cura 2.6.