Ultimaker/Cura #1259: Cura 2.3.1 nozzle size est time

ChrisRiddell, 2016-12-20:
Not sure if this is a bug or not but seems pretty odd.

When you have a larger nozzle size selected the print time estimate is longer then that of a smaller nozzle on 0.15mm layer height. as you can see in these screenshots.

[image]
[image]

also out of curiosity why don't the 0.6,0.8 nozzle sizes have normal and fast quality settings?

Ghostkeeper, 2016-12-20:
It depends a little bit on your model, but usually the larger nozzle sizes have a much shorter print time. Take for instance a 4x4x4cm cube:
* 0.25mm nozzle: 16h 21m
* 0.4mm nozzle: 7h 16m
* 0.6mm nozzle: 2h 23m
* 0.8mm nozzle: 1h 32m

If you have a print with lots of small pieces and you try to print it with a large nozzle, it could in theory be slower, but you'd have to have a pretty extreme example. Larger nozzle sizes have a greater layer height by default and a bigger line width, and these two are the greatest factors in the print time.

For 0.6 and 0.8 we had only one profile made per material. This is because there would otherwise be way too many profiles to make, for every combination of nozzle size, material and quality level. For 0.8 we decided to call it "Fast" because it's not meant to be any typical quality.

ChrisRiddell, 2016-12-20:
@Ghostkeeper I used the file  small-planter-reservoir.STL from http://www.thingiverse.com/thing:903411

and got the time estimate shown in them screenshots, would this fall into the small details?

also thanks for for taking the time to give all this info.

Ghostkeeper, 2016-12-20:
Ah, but I hadn't read your question well enough. You're comparing "Fast" at 0.4mm with "Normal" at 0.6mm. Not "Normal" at 0.4mm.

I'm measuring these estimates:
* 0.25mm nozzle: 2d 22h 34m
* 0.4mm nozzle: 19h 50m (normal), 9h 44m (fast)
* 0.6mm nozzle: 9h 51m
* 0.8mm nozzle: 6h 10m

So I'm seeing the same: 9h 44m for Fast on 0.4mm, 9h 51m for Normal on 0.6mm.

At 0.4mm/Fast, the wall speeds are 40/50 and top/bottom speed is 30. At 0.6mm/Normal, the wall speeds are 25/40 and top/bottom speed is 20. Since your print is like 90% wall and for the rest skin, this has a huge impact on your total time estimate.

You can try to set the outer wall speed a bit higher to 40mm/s too. That should shave off like 20% of the time right there. If you want to increase it any more than that you will also have to increase the temperature. Eventually all that filament just won't have time to melt any more. It needs at least 4 seconds in the heat zone for the heat to reach the core of the 2.85mm filament.

But in short: It is expected that "Fast" is faster than you'd think for a 0.4mm nozzle, when comparing it with a 0.6mm nozzle.

I still think all of the profiles are on the slow side... But you know, they want to show off with the pretty pictures of prints...

closed 2016-12-20 by Ghostkeeper