Ultimaker/Cura #1262: Not One Piece

SolidWorksMagi, 2016-12-20:
I have a design printing but the components do not connect into one large part in the print ???

[image]

nallath, 2016-12-20:
Could you view the model in the x-ray view? I think your model is broken. If there are red areas in the x-ray view, it means that this is the case.

SolidWorksMagi, 2016-12-20:
Hi, 

 

The model .STL file is from SolidWorks … 

 

I actually noticed this problem with other things where components cross each other so CURA makes spaces in those areas rather than seeing them as a solid.  

 

This is from making separate parts in SolidWorks, then assembling them and saving the assembly as a solid part, then saving the solid part as a .STL file.  

 

I've been creating a new crisscross all from one 3D SolidWorks part to see if the mesh becomes one better that way than assembly conversions.  

 

This is weird because I've never had this problem with other GCode generators.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

Ghostkeeper, 2016-12-21:
Is there any way in which we can reproduce this?

I don't have SolidWorks, but when I create a piece in OpenSCAD that crosses its own components it doesn't have this problem.

All bets are off when there are internal walls, because it's difficult to determine then what the inside of the component is. In this case, the even-odd rule would make the inside hollow.

SolidWorksMagi, 2016-12-21:
Hi, 

 

What comes from SolidWorks is a solid.  The slicer turns components hollow.  What it looks like is that the slicer is seeing the individual meshes and turning them into its own separate component again.  

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

Ghostkeeper, 2016-12-22:
How could it see the individual meshes if it is a single solid?

Is there a way I can reproduce this? Perhaps you have a file that we can test with?

SolidWorksMagi, 2016-12-22:
Hi, 

 

Here's one of the files I was using and getting those odd results;  

http://www.brainless.org/OtherStuff/HelicalVase10.STL 

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

Ando120, 2016-12-22:
There object crosses its own components.

I get that result only when I turn off option: Mesh Fixes > Union Overlaping Volumes.
By default its on.

SolidWorksMagi, 2016-12-22:
Hi, 

 

So I should have a checkmark on:  "Mesh Fixes > Union Overlaping Volumes"? 

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

Ando120, 2016-12-22:
Yes, then it looks fine in layers view.

Ghostkeeper, 2016-12-23:
Looking at the file in X-Ray View, as Nallath said, shows that the mesh is not closed:
[image]

X-Ray View colours everything red where there are an odd number of surfaces behind each other. This means that SolidWorks must've produced meshes that are not water tight or have additional non-water-tight shapes inside the outer shell.

Apart from the Union Overlapping Volumes option, you could also try "Extensive Stitching" under Mesh Fixes.

Ghostkeeper, 2016-12-23:
It seems to slice fine with default settings in 2.4, by the way. No subtracting from itself.

closed 2017-07-27 by nallath