Ultimaker/Cura #1263: Cura 2.3.1-0 Enable Retraction Causes Massive Retraction at the End of Print

Lasakro, 2016-12-21:
Enabling this causes randomly lengthy retraction to be added to the g-code. Sample below of the end of a print directly after TIME_ELAPSED.

> ...
...
...
G0 F7200 X101.93 Y136.506
G1 F600 X101.506 Y136.929 E86.46233
G0 F7200 X101.506 Y136.6
;TIME_ELAPSED:180.797658
**G1 F1500 E79.96233**
M107
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = empty\\ndefiniti
;SETTING_3 on = prusa_i3_xl\\n\\n[metadata]\\ntype = quality_changes\\nquality_t
;SETTING_3 ype = normal\\n\\n[values]\\n\\n"}
/>

The balloon pop-up states enabling this is for retraction for on-print moves.

Ghostkeeper, 2016-12-21:
It's retracting 6.5mm. That's the default value for retraction.

Do you think it is more appropriate to use the Switch Extruder Retraction Amount?

Lasakro, 2016-12-21:
Maybe. All I know is that after one print, I don't remember which one, my filament was completely unloaded from my 70cm tube. After writing the above I now realize that the end of print head move is being treated as a non-print move so unsure. Maybe my system got confused so a bit more testing is needed and see if it persists. Below is from 2 other larger prints:

;TIME_ELAPSED:1189.905230
G1 F2400 E1690.15523

;TIME_ELAPSED:28647.670740
G1 F2400 E3674.56328

If it's normal this issue should be closed.

Ghostkeeper, 2016-12-22:
@BagelOrb do you know about the move or retraction at the end of the print?

Lasakro, 2016-12-22:
I made 2 more prints last night and both got to the end of the print, left the head hot on the print then started something like a 10 minute long filament unload. After that the print actually finished by moving the head off the print and Duet Wifi Web reported that the print was finished. Somethings hooky going on here.

Lasakro, 2016-12-24:
Solved: There is a subtle difference between Marlin and Reprap firmwares. Cura doesn't take this into account and shouldn't lump the 2 together. The default end code requires a M83 after the M91. RepRapFirmware latest revision firmware uses M83 to set the extruder to relative mode: extrusion is NOT set to relative by ReprapFirmware on G91: only X,Y and Z are set to relative. By contrast, Marlin (for example) DOES also set extrusion to relative on a G91 command, as well as setting X, Y and Z.

fieldOfView, 2016-12-24:
In the Machine Settings for Cura 2.4, you can now select "Repetier" as the GCode flavor. That might fix this, especially if it worked before in legacy Cura.

Lasakro, 2016-12-24:
Repetier has nothing to do with this issue.

fieldOfView, 2016-12-24:
Derp, I misread your message as Marlin vs Repetier instead of Marlin vs Reprap.

rexipus, 2016-12-30:
I believe that this is because your printer is not recognizing the G91 gcode for setting relative positioning for the extruder axis. My DuetWifi also does this. To fix it, go into your gcode and insert an M83 after your G91.

If your printer doesn't apply the G91 to the extruder axis, then this code that you posted will indeed do a very long retraction at the end:

"G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more"

Notice the E-1 and the E-5. If G91 isn't applied to the extruder axis, and you had extruded out to E86.46233 by the end of your file, then the E-1 will attempt to unwind the filament all the way from 86.45233 back to -1, which is actually more filament even than you used during the print. 

Try the M83 after the G91 and see if that works, though. That will allow the E-1 and E-5 to be treated as relative moves, which is what's intended by that code.

Lasakro, 2016-12-30:
Agreed. See post from 7 days ago.

rexipus, 2016-12-30:
Doh! Sorry.

Lasakro, 2017-01-02:
@Ghostkeeper,
For configuration control purposes shouldn't your tracking # posted here and this issue closed in Github?

Ghostkeeper, 2017-01-03:
It's not really necessary. The Jira issue number is CURA-3188 but it's linked to this Github issue already. Ultimaker wants to keep the Jira tracker closed to public though so it can plan things that aren't supposed to be public yet.

Typically we keep the Github post open until it's all fixed so that the public knows we're still working on it. Then typically we also forget to close the Github issue afterwards, but that's another matter.

BagelOrb, 2017-01-06:
Thanks for linjking to the JIRA issue, @Ghostkeeper :)

If I understand correctly then Marling should ignore M83 when already in G91 mode, so we could simply always add the M82 after a G90 and always add an M83 after a G91. Is that correct @daid ?

daid, 2017-01-06:
Relative modes are a pain in the ass, try to avoid them, as I cannot guarantee that the UM3 does these correct as well.

M83 is a custom extension which might not be supported by all firmwares. The mixing of G90/G91 and M83 could also have different behavior on different firmwares.

bubnikv, 2017-01-06:
By the way, the Prusa i3 MK2 runs a somehow modified 1 year old Marlin
firmware with G-codes generated by the Slic3r with relative E axis without
any problem. The only problem we experienced is a loss of the accuracy of
the firmware E counter if the E axis does not get zeroed for multiple
prints. Calling G92 E0 at the start of a print is sufficient to keep the
floating point counter of the E axis accurate enough.

BagelOrb, 2017-01-10:
The `G91` is in the end gcode of your machine. you should edit yourmachine.def.json.

closed 2017-01-10 by BagelOrb