Ultimaker/Cura #1264: works X axis only

InfNull, 2016-12-22:
rotate, move, scale, mirror avaliable X axis only. I can't select green and bluse axes by mouse.

Ghostkeeper, 2016-12-23:
Cannot reproduce. Perhaps it works better if you rotate the camera a bit?

If not, what graphics card and OpenGL version do you have? I can help you obtain that information if I know what operating system you're using.

InfNull, 2016-12-25:
Camera rotation does not help.
I have ATI Mobility Radeon HD 4570 Graphics card and 3.1 OpenGL version.
Just in case i use Windows 7 maximum x64

Ghostkeeper, 2016-12-27:
I've made an issue about it in our internal tracker. We'll take a look to fix it in 2017 if it can be reproduced on our Windows machines.

jackha, 2017-01-18:
I cannot reproduce it either.
- What Cura version are you using?
- What exactly happens when you select a model and hover the mouse over the arrows? Are they lighting up?
- What exactly happens if you click and drag? Is it like a misclick?

One thing that I can think of is that when you click near the center of the axis, it will not react. This can happen on small objects of 2 cm or so.

InfNull, 2017-01-18:
I'm using Cura 2.3.1, and also tried 2.4.0 beta.
when I select a model and hover the mouse over the arrows - they are not lighting up.
And if I click and drag - it looks like a misclick, model deselect.

I tried click everywhere (near the center of axis, far of center, tried rotate on different angles) - nothing. 
Translate (Move) tool have same behavior. My objects generally bigger than 2 cm.

Only red axis works.

Ghostkeeper, 2017-01-22:
Hmm, could be a problem with the selection render pass. We render the scene in a special buffer with special colours (pure red for X, pure green for Y and pure blue for Z, I believe). Which object is hovered by the mouse cursor is determined based on the colour of the pixel below the mouse cursor on that buffer.

Ghostkeeper, 2017-05-01:
We still haven't been able to reproduce this. After 3 months of inactivity, our product manager closed the internal issue.

closed 2017-11-14 by ChrisTerBeke