Ultimaker/Cura #1268: [2.4] Print settings not been followed

ChrisRiddell, 2016-12-24:
Hello,

Printer: Ultimaker 2+
Cura: 2.4 Beta 1

I tried printing the file ARROW-LOADER.STL from http://www.thingiverse.com/thing:1848665

(Note: this print came out alot better under 2.3.1)

The print failed as the speed was alot faster (like I thought the print head was going to take off) then what was set and the cooling fan never came on even though it was checked, you will see it failed to do a fill over supports and some of the supports failed to print.

Here is a photo

[image]

I copied the CPE material profile and changed the temp, retraction dist and speed and set the cooling fan speed to 50%.

Here is a video of how I set up cura as well https://youtu.be/oPDdO3yqcPY

(Note: I have 50% fan and the right temps set on my customised material on the ultimaker its self)

here is the gcode as well
[UM2_ARROW-LOADER.gcode.zip](https://github.com/Ultimaker/Cura/files/671986/UM2_ARROW-LOADER.gcode.zip)

Ghostkeeper, 2016-12-24:
The speed in your g-code file looks normal. I see F7200 for travel moves (120mm/s) and the highest extrusion move I found was F1800 (30mm/s) for inner walls. (There is no infill.) There are also no fan commands in the g-code. I believe that is controlled by the firmware as well.

You didn't happen to have set the print speed higher on the printer, right? While printing, in the menu: Tune -> Speed.

ChrisRiddell, 2016-12-25:
I checked the print speed under tune -> speed its 100%

I also tried a couple different qualities still getting the same results of the photo above.

Not sure why the fans don't come on though.

ChrisRiddell, 2016-12-25:
Okay I deleted the config folder and reloaded Cura as i had cura 2.3 installed before it will now not let me slice it with the fan set at 50% only at 100%.

Ghostkeeper, 2016-12-27:
Uh, what happens when you try to set it to 50%? Does the setting become red?

ChrisRiddell, 2016-12-28:
Setting does not become red
[image]

But set at 100% the ultimakers material profile overrides it anyways so it's a little confusing as you would think the cura settings would override the printers?

Ghostkeeper, 2016-12-28:
Ah, I see what's happening. The Maximum Fan Speed is lower than the Regular Fan Speed. The problem is that Regular Fan Speed doesn't scale along with the Fan Speed parent setting, because it's been given a static value of 80 by the profile. Maximum Fan Speed (which is hidden in your screenshot) does still scale with Fan Speed, so it becomes 50.

If you were to make the Maximum Fan Speed setting visible, you'd see that it is red.

I've adjusted the UM2+ profiles such that Regular Fan Speed scales with Fan Speed, so people won't have this problem any more with this particular setting. This won't make it for release 2.4 though. It'll be in 2.5 and onwards. See commit https://github.com/Ultimaker/Cura/commit/ae170830fb7c94e5acb5e4d19b2782ddb15cbfd7 .

closed 2016-12-28 by Ghostkeeper