Ultimaker/Cura #1273: Material Setting for Filament Diameter not recognized

1-prop-head, 2016-12-27:
Ultimaker 2 Ex (not plus),Original Filament Feeder,1.75mm ABS filament,Cura 2.4 Beta
Printer filament setting is ABS 1.75
Printer is as above.  When printing I need to set the Meterial Flow to 55% to be able to not grosely over extruding the filament

Ghostkeeper, 2016-12-28:
Hmm, you'd expect the flow to be reduced if no compensation were applied. Perhaps we're compensating double?

1-prop-head, 2017-05-27:
This is not closed in my opinion.  I Just loaded 2.5.0 (latest firmware 2.4ex) and with my UM2ex I still need to change material flow from 100% to 22% to not over extrude 1.75 abs filament. This was not the case with the firmware provided with the old version of cure (15...) which extrudes correctly at the default 100%

Ghostkeeper, 2017-11-23:
This was fixed via https://github.com/Ultimaker/Cura/pull/1890 in Cura 2.6, but it doesn't show as fixed because the commits were cherry-picked into the release.

1-prop-head, 2017-11-24:
What was the date on this and no it is not fixed 

Fixed or not in CURA 3.04 and all previous iterations I must use a material speed adjustment in the printer of 57% to not over extrude.

Ghostkeeper, 2017-11-29:
I think I misunderstood your initial question, sorry. I was thinking about the UM2E+ which has material profiles. The fix allowed the diameter in the material manager to be used for slicing. The UM2E has no material profiles so the fix in question did not apply.

Cura has no knowledge of what material is in that printer, be it 1.75mm or 2.85mm. It'll assume 2.85mm then, since that is the only option in the firmware. If you're slicing for a different diameter than the firmware thinks it has, I'd say that Flow is a fine setting to adjust. Since the E parameters for UltiGCode indicates filament length, not volume, you should be changing the material flow to 265% (=2.85^2 / 1.75^2) in theory. Why you had to compensate in the other direction I don't know.

If you think that the firmware needs to be adjusted to allow for 1.75mm diameter materials, you're looking for the repository of the [firmware of the UM2](https://github.com/Ultimaker/Ultimaker2Marlin), I'd say?

fieldOfView, 2017-11-29:
There should not be a material diameter field in Cura at all if you are using UltiGcode with the Ultimaker 2, because in that case Cura should be material-agnostic. 

I think you can get this to work properly by switching from UltiGcode flavor (labeled "Ultimaker 2" in Cura) to Marlin flavor. Then you get full control over all material settings in Cura, including diameter. Also Cura will no longer use volumetric extrusion, so the E extrusion values should be correct.

alekseisasin, 2018-05-08:
@1-prop-head , do you still have a difficulty with filament diameter? Or did you try to change flavors?