Ultimaker/Cura #1275: Cura / CuraEngine heats non-existent bed

Ocarthon, 2016-12-28:
I have wrote a custom configuration for my Vellemann K8400. It worked fine until yesterday when i updated Cura to the current 2.4 Beta. The configuration sets `machine_heated_bed` and `material_bed_temp_prepend` to false, but in the generated GCode, a M140 command is inserted, resulting in my printer to stop due to a heater error.

Ghostkeeper, 2016-12-29:
Duplicate of #1209.

fieldOfView, 2016-12-29:
@Ghostkeeper, either you linked to the wrong issue, or you misread this issue.

@Ocarthon. could you paste your custom configuration into pastebin or into a github gist?

Ocarthon, 2016-12-29:
[k8400.def.json](https://gist.github.com/Ocarthon/a6f26b1b33cbdb27580145aabbd01a7c)
[k8400_extruder_1.def.json](https://gist.github.com/Ocarthon/29ddd6f665a8703610356f9214627434)
[k8400_extruder_2.def.json](https://gist.github.com/Ocarthon/9bc0cece4834c0f2ccd6105d9cdff93b)

At least at some point Cura does read the configuration correct, because i cannot change the bed temperature even if i make that setting visible.

Ghostkeeper, 2016-12-29:
I think I misread it. It's not about disabling M140 via setting the temperature to 0 but about disabling M140 via setting the enabled-setting to false.

Ocarthon, 2016-12-29:
In the meantime i looked a bit in the code and probably found the solution. I create a seperate [issue](https://github.com/Ultimaker/CuraEngine/issues/428) in the CuraEngine repo.

BagelOrb, 2017-07-23:
fixed

closed 2017-07-23 by BagelOrb