Ultimaker/Cura #1276: Cura can't find UM2E+ device via USB

Ghostkeeper, 2016-12-29:
I recently upgraded my UM2 Extended to a UM2 Extended+. Nice, strong feeder!

However when it came to upgrading my firmware I stumbled on a problem: Cura can't seem to connect to this device. I'm using the Cura 2.4 beta, the AppImage, on Ubuntu 16.04. This is the error I'm getting:

    2016-12-29 03:11:41,480 - DEBUG - USBPrinting.USBPrinterOutputDeviceManager.getDefaultFirmwareName [196]: Choosing basic firmware for machine ultimaker2_extended_plus.
    2016-12-29 03:11:43,615 - INFO - USBPrinting.USBPrinterOutputDevice.updateFirmware [265]: Updating firmware of /dev/ttyACM0 using /tmp/.mount_2cJ5Dl/usr/bin/resources/firmware/MarlinUltimaker2extended-plus.hex
    2016-12-29 03:11:44,915 - ERROR - USBPrinting.USBPrinterOutputDevice._updateFirmware [201]: Unable to connect with serial. Could not update firmware
    2016-12-29 03:11:44,918 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [298]: Attempting to connect to /dev/ttyACM0
    2016-12-29 03:11:44,922 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/ttyACM0 to remove
    2016-12-29 03:11:44,923 - INFO - USBPrinting.USBPrinterOutputDevice._connect [305]: Could not establish connection on /dev/ttyACM0: 'Failed to open serial port'. Device is not arduino based.
    2016-12-29 03:11:44,924 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
    2016-12-29 03:11:44,925 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,925 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 250000
    2016-12-29 03:11:44,926 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,926 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 230400
    2016-12-29 03:11:44,927 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,928 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 57600
    2016-12-29 03:11:44,928 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,929 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 38400
    2016-12-29 03:11:44,930 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,931 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 19200
    2016-12-29 03:11:44,931 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,932 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 9600
    2016-12-29 03:11:44,932 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port /dev/ttyACM0
    2016-12-29 03:11:44,933 - ERROR - USBPrinting.USBPrinterOutputDevice._connect [350]: Baud rate detection for /dev/ttyACM0 failed
    2016-12-29 03:11:44,933 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [364]: Closing the USB printer connection.
    2016-12-29 03:11:44,934 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [369]: PrinterConnection.close: cannot join current thread (expected)
    2016-12-29 03:11:44,948 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/ttyACM0 to remove

When I plug the cable in, the file /dev/ttyACM0 is popping up.

Right now I can't change the firmware via Cura so the printer still thinks it's a UM2E (without +).

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-24:
It doesn't reproduce any more, so apparently not. Cura now connected to my printer and a successful firmware upgrade was made.

closed 2017-11-24 by Ghostkeeper