Ultimaker/Cura #1277: Request for official Cura support channel on FreeNode IRC network

Patola, 2016-12-29:
Hi,

I and others have been very interested in Cura usage and development, and in a number of scattered forums have been asking and helping other users with nothing but unabashed love for this fantastic piece of software.

However, as Cura gets more and more sophisticated, it is obvious that these scattering is not positive. I have seen dozens of request for help from facebook groups, telegram group, IRC channels for other software (#freecad, #slic3r etc.) but there is no central point for helping cura users.

FreeNode is a traditional IRC messaging network where people gather to discuss development of open-source projects and help people. They have an explicit policy of partnership with open-source projects like cura, and unofficial channels start with ## while official channel have a single # on its name.

The #cura channel has its name reserved by freenode staff, being so important a project:
```
[07:39] [ptl<+Zi>] [1:freenode ] [Act: 1,3,4,5,6,7,8,9,10,17]                       
[(status)] /chanserv info #cura
07:32 -ChanServ(ChanServ@services.)- Information on #cura:
07:32 -ChanServ(ChanServ@services.)- Founder    : freenode-staff
07:32 -ChanServ(ChanServ@services.)- Registered : Nov 29 19:13:45 2006 (10y 4w 4d ago)
07:32 -ChanServ(ChanServ@services.)- Last used  : Dec 11 08:21:57 2006 (10y 3w 0d ago)
07:32 -ChanServ(ChanServ@services.)- Mode lock  : +ns-t
07:32 -ChanServ(ChanServ@services.)- *** End of Info ***
```

The FreeNode policy for official channels and groups is there: http://freenode.net/groupreg

My suggestion is that we get official word from this group - maybe from some very active members which are already on freenode like @Ghostkeeper and @thopiekar - and get the #cura channel officially registered and populated. I am no active member, but I would love to collaborate and stay there to discuss issues and help people -- and of course, try and get other people to help. I think the public would rapidly increase due to interest of cura. This would require changing the README.md to inform people about this way of getting support and help.

Patola, 2016-12-29:
@RafaelEstevamReis this might interest you

thopiekar, 2016-12-29:
Well, I think the channel itself is no problem, but getting people connected to IRC is more difficult.
Most people prefer forums because you can access them easily with a browser.
I never liked these applets and prefer native clients.

Btw. Ultimaker has a chat in their community section on their website :wink: 

I personally join this channel for a long time since I noticed that 1 or 2 people are online there very often.

nallath, 2016-12-29:
We already have the cura-dev chanel on freenode. Same goes for  the ultimaker channel.

I wouldn't mind having a cura channel, but my IRC-fu is a bit lacking (read nonexistent)

Patola, 2016-12-29:
@thopiekar there are already some type of people that prefer to get in IRC, not necessarily developers. I think the important thing would be to add #cura to the README.md which would make github users (hopefully more technically inclined) get to it. Maybe #cura-dev also - where was this one documented? I didn't see it anywhere.

Anyway, according to the freenode documentation, we need some kind of 'official' endorsement for the channel, is there any way to get Ultimaker people invoved in this? Or, if official relationship with the freenode network is not desirable nor possible, would it be sensible to at least add ##cura to the README.md?

Ghostkeeper, 2016-12-29:
#cura-dev is not documented anywhere as far as I know. It started as a way for Cura developers and contributors to talk to each other in deep technical details.

Ghostkeeper, 2016-12-29:
As an aside, the actual Ultimaker IRC channel is quiet enough that it doesn't hurt to just discuss Cura stuff there ;)

nallath, 2016-12-30:
On the other hand, there are people that use cura that are not directly associated with Ultimaker.

thopiekar, 2016-12-30:
If you like I can re-register for this channel (currently the freenode-staff holds the ownership).
I just had a short discussion about the registering process at #freenode.

Patola, 2016-12-31:
I would surely like it. I do not want to take from the ultimaker chat, I think the public that would go for IRC support is a completely different public. And btw, I do think it is also important to mention the Ultimaker chat address on the README.md for Cura. Having a place for quick questions is important for several types of inquisitive users. Also, sometimes quickly resolving small doubts in chats can prevent frivolous bug reports.

awhiemstra, 2017-01-05:
As someone who uses IRC quite a bit I would like to have #cura. In fact I originally intended to use #cura instead of #cura-dev but then noticed it was already registered. If we can get #cura I would actually prefer to close down #cura-dev unless we expect a lot of traffic to #cura. 

In regards to mixing user support and development chat, another channel I'm in (#krita) is also a mix of user support and development and that seems to work quite well. Especially since the division between user support and development is rather vague when you consider things like creating a new JSON definition.

Ghostkeeper, 2017-01-05:
Right, so I think we agree then.

Plan of action:
* Request ownership over #cura.
* Document the #cura IRC channel here at Github.

thopiekar, 2017-01-05:
Already sent a requested for the ownership on 30.12.2016, but didn't get any answer so far.

thopiekar, 2017-01-06:
Asked again on #freenode why there was no reply so far. Due to the holidays they've got a lot of mails to be processed. I guess it is a question of time until we get an answer here.

probonopd, 2017-01-07:
Just curious, why don't people just join #cura as-it-is? Seems to be working but only 2 people in there right now.

closed 2017-11-14 by ChrisTerBeke

Ghostkeeper, 2017-11-23:
I've documented the channel in the README file.