Ultimaker/Cura #1281: [Cura 2.1.3 with Prusa i3 clone]Z-ax-movement into object at print end

ManfredEP, 2016-12-30:
At the end of the print of two test objects (OpenSCAD-cubes without any bottom/top/fill) the z-ax is moved 2 miilmeters down (into the object) before any movement to side end. Never before. Now retraction is enabled what was not before. Test without retraction follows. 

Photo shows the objects. In black one the damage is hard to see but in green one the black adhesions (from outside pollution of nozzle during black item print) make it easy to see. 

[image]

nallath, 2017-01-02:
Could you also print the object with the same material twice? The color can have a pretty big impact on the behavior of the material.

Ghostkeeper, 2017-01-02:
I'm more interested in the g-code, to see whether it actually moved 2mm down or whether it was the printer. I can't imagine it moving 2mm down at the end of the print on purpose, unless the Z Hop Height is negative or something...

ManfredEP, 2017-01-03:
Hello,
the g-code can be found here. I don't want to load it here directly because it's about 1.77 MByte in size. I wasn't sure this will work here.
[www.preussig-ferien.eu/Parkplatz/ABS_Test_R=20x10+R=20x40.gcode](url)
Please direct download since the dir is blocked so a file list or search is not possible. 

This is the complete file without any change. So the parts are made. 

greetings

Manni

Ghostkeeper, 2017-01-04:
The g-code shows no such move. There are two moves towards the end where the Z-coordinate is changed. First there's the move to the layer height of the last layer:

    ;LAYER:397
    G0 X121.375 Y98.116 Z39.999

Apart from the apparent rounding error, this is intended. The previous Z coordinate was 39.899.

Then there is a retract and move in the ending procedure:

    G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more

This is a relative move and could be the culprit. But the Z coordinate is increased by 0.5, not decreased. If this crashes into the model that is a bug in the printer. Perhaps you could try changing that line into this:

G1 Z-0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more

I've flipped the sign of the Z coordinate. If that fixes things, then we should modify the end g-code setting for your printer type to account for this bug.

ManfredEP, 2017-01-04:
Thanks. I will try this. The firmware has a few more problems (f.e. is z to short. 9.6 mm at 10 chosen) or possible bugs so this is allready possible. Update don't work since serial is offline. There is a new mainboard on tour but from china this lasts some time. Especially since customs seems to check each sending to me I don't know if it's so and when why but every time I got parts from asia this last a lot of time due to customs. So this work-around can possibly be very helpful.

greetings 
Manni

ManfredEP, 2017-01-04:
Hello,
@Ghostkeeper: an addition: I've just looking for the line as you mentioned. As I just change the sign I suddenly remember that the shift down was static in x and y-axis. So this command you have found could not be the reason for the problem since there is a movement in both other directions in the same command. So I looked for other points before and find this just a little bit above: 
[image]
The printer is in absolute mode. I don't know why but I think this is the problem. For any unknown reason the real position don't match this values and so the nozzle comes down to much. I check this by deleting this line and than give you a report. 

greetings
Manni

closed 2017-11-14 by ChrisTerBeke