Ultimaker/Cura #1282: Cura 15.4.6 for Mac, impossible to choose custom machine

kennywyland, 2016-12-30:
I'm using 15.x because the maker of my 3d printer (monoprice 15365) suggested 15.x instead of the new 2.x because 2.x was asking for configuration options that weren't in the manual for the printer.

It's not possible to choose a Custom Machine because the window is too small, doesn't scroll, and isn't resizable. The list is cut off just below "julia".

Patola, 2016-12-31:
I am not an official developer - just another user - but if you have a bug with the older Cura, you have to open a bug report in the correct github repository, legacy Cura: https://github.com/daid/LegacyCura

As an user to another, though, I would highly recommend the new Cura, not only because it would not have the bug you mentioned, but because it is vastly improved over the old version. And frankly, I think that giving up an excellent piece of software just because some settings or words differ from your reference material is quite depressing.

nallath, 2017-01-02:
You can report it in legacy Cura, but don't expect anyone to fix it.

Ghostkeeper, 2017-01-02:
Perhaps keyboard navigation is a workaround.

Ultimaker is no longer maintaining legacy Cura since release 2.1. It's up to pull requests from the community to fix any remaining issues, really.

closed 2017-07-23 by BagelOrb