Ultimaker/Cura #1283: 2.4: Crash on launch / Settings lost

CCS86, 2016-12-31:
Hi guys,

I ran cura 2.4 beta last night and it worked perfectly. It saw my existing machines and print settings automatically. I tweaked a few things, output some code, and closed it. 

Tonight, I launched it and just after clearing the loading window, there was a brief glimpse of the main window, then a windows "cura has stopped working...". I tried again and the same thing happened.

Then I went to 2.3.1, it loaded up and asked me to creat a machine. wth. No machines or profiles to be seen. :/

OS is Windows 10.

CCS86, 2016-12-31:
Log entries from this: 

`2016-12-30 00:45:42,708 - DEBUG - UM.Qt.QtApplication.windowClosed [198]: Shutting down cura
2016-12-30 00:45:47,404 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2016-12-30 00:45:47,405 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [203]: Killing engine process
2016-12-30 00:45:47,408 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [206]: Engine process is killed. Received return code 1
2016-12-30 19:27:36,180 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2016-12-30 19:27:36,187 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2016-12-30 19:27:36,195 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2016-12-30 19:27:36,204 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2016-12-30 19:27:36,217 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2016-12-30 19:27:36,225 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2016-12-30 19:27:36,234 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2016-12-30 19:27:36,242 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2016-12-30 19:27:36,252 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2016-12-30 19:27:36,262 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PostProcessingPlugin
2016-12-30 19:27:36,277 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2016-12-30 19:27:36,290 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2016-12-30 19:27:36,299 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2016-12-30 19:27:36,308 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2016-12-30 19:27:36,324 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2016-12-30 19:27:36,338 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2016-12-30 19:27:36,352 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2016-12-30 19:27:36,361 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2016-12-30 19:27:36,371 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2016-12-30 19:27:36,388 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2016-12-30 19:27:36,449 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2016-12-30 19:27:36,506 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UpdateChecker
2016-12-30 19:27:36,508 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2016-12-30 19:27:36,547 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2016-12-30 19:27:36,562 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2016-12-30 19:27:36,570 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2016-12-30 19:27:36,578 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2016-12-30 19:27:36,587 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2016-12-30 19:27:36,596 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2016-12-30 19:27:36,610 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2016-12-30 19:27:36,620 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2016-12-30 19:27:36,622 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'external-backend': False, 'file': [], 'disable-textures': False}
2016-12-30 19:27:36,633 - INFO - UM.VersionUpgradeManager.upgrade [99]: Looking for old configuration files to upgrade.
2016-12-30 19:27:36,685 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: 'NoneType' object is not iterable
2016-12-30 19:27:36,686 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
2016-12-30 19:27:36,687 - WARNING - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 305, in _upgradeFile
2016-12-30 19:27:36,689 - WARNING - UM.Logger.logException [69]: TypeError: 'NoneType' object is not iterable
2016-12-30 19:27:36,694 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2016-12-30 19:27:36,782 - INFO - UM.Backend.Backend.startEngine [71]: Started engine process: C:\Program Files\Cura 2.4-beta\CuraEngine.exe
2016-12-30 19:27:36,785 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.4-beta\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.4-beta\\resources\\definitions\\fdmprinter.def.json', '']
2016-12-30 19:27:36,896 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2016-12-30 19:27:36,900 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2016-12-30 19:27:36,903 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2016-12-30 19:27:36,905 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2016-12-30 19:27:36,906 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
[... cut ...]
2016-12-30 19:28:46,464 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py (windowClosed [197]): Shutting down cura
2016-12-30 19:28:48,353 - DEBUG - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Attempting to kill the engine process
2016-12-30 19:28:48,353 - DEBUG - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Killing engine process
2016-12-30 19:28:48,357 - DEBUG - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Engine process is killed. Received return code 1`

nallath, 2017-01-02:
It should have saved your old settings in a seperate folder called "old". Could you have a look if this happend?

Ghostkeeper, 2017-01-02:
2016-12-30 19:27:55,857 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: 'NoneType' object is not iterable
    2016-12-30 19:27:55,858 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
    2016-12-30 19:27:55,859 - WARNING - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 305, in _upgradeFile
    2016-12-30 19:27:55,860 - WARNING - UM.Logger.logException [69]: TypeError: 'NoneType' object is not iterable

Looks like this is the problem. I suspect that the version upgrade plug-in returned `None` somewhere where this is unexpected.

lolorc, 2017-01-03:
did it to me as well, it's well annoying.
upgraded from next (20161220) to today. I've lost everything, don't know how to import tons of old setting back

MisterTazUSA, 2017-01-03:
Hello,

It seems I have the same issue. I loaded 2.4 and ran it, restarted the application the next day and get similar results as above; splash screen appears and then nothing.  I have not installed another version of CURA, this is a virgin install on a new machine running Windows 10?  Like I said after initial install the application ran but will no longer start after closing, any ideas would be greatly appreciated?

[stderr.log.txt](https://github.com/Ultimaker/Cura/files/682741/stderr.log.txt)

CCS86, 2017-01-06:
I always install new versions to a new directory. But for some time, different versions have shared common appdata.

CCS86, 2017-01-08:
@nallath "It should have saved your old settings in a seperate folder called "old". Could you have a look if this happend?"

The only thing in "old" is a cura.cfg, without much data.

I has set everything back up and run 2.4 a few times successfully. Now it just did it again, and lost my machine and print settings :(

No Windows crash first this time though.

nallath, 2017-01-09:
Every time it sees a profile with the wrong version, it moves it to the old folder & tries to upgrade. 

I'm just saying that the values that you used aren't gone, the automatic updater doesn't like it. Could you post your backup somewhere? Then one of us can test & see what is going on.

CCS86, 2017-01-09:
This is all that's in the old\3 directory.
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/693685/cura.cfg.txt)

nallath, 2017-01-10:
Huh, that's weird. It looks like it only tried to upgrade the preferences file.

ianpaschal, 2018-02-21:
I'm closing this since it's over one year old and this behavior has been reworked recently.

closed 2018-02-21 by ianpaschal