Ultimaker/Cura #1284: Possible Improvement: Solid layer infill for "free hanging" features inside parts or for support structures

Taiko1976, 2016-12-31:
Hi there,

I've switched to Cura lately and setup my RenkForce RF2000 which works neatly with your latest Version of Cura (btw.: anyone who wants to setup an unsuppoted printer in Cura: Just type in your printers size and the start-/end codes of your previously used, working slicer... worked as a charm for me). Nice work! ... keep it up!

Background:
I'm working on an RC airplane that will be completey produced by 3d printing. For this I need my parts to be ultra light which translates into parts with a minimum infill (e.g. 5%).

Issue:
Obviously these parts will be complex (e.g. tubing running trough parts, cut outs with edges parallel to print layer, etc.). Therefore it can happen that a feature inside the part starts free hanging (printing in the air) or an edge is free hanging (not supported by the infill) which will result in holes in the skin.

Proposed solution:
Whenever a feature is "free hanging" inbetween the infill fill the corresponding chamber(s) with 1 or 2 solid layers to give the feature "solid ground" before the feature itself is started. Number of layers could be configurable. This could also be used for support structures which would greatly reduce the ammount of filament used for support structures and hence improve cost performance of your product.

Best regards,
Bernd

nallath, 2017-01-02:
I've added bagelOrb (him being "the engine guy") to this issue so he can chime in.

Ghostkeeper, 2017-01-02:
In version 2.4 we have an experimental feature that is completely different from what you're proposing but could also remedy the problem you're having: "Hollow Out Object". This removes all infill, but then makes the insides of the object eligible for support. So those features could then be supported by whatever the support generation would deem good. If you then set the support overhang angle and the support X/Y distance very low, you'll get sort of a similar result. It's not the same though, and particularly the walls will be much weaker and the structure will be heavier for the same amount of strength. So it's probably not the end result you want.

What would also work well is to set the infill much higher (like 50%) and then select the new Cubic Subdivision infill. Also a 2.4 feature. This makes the infill in the centre very sparse but at the sides very dense, up to your selected 50%. Because of the dense infill at the sides and top/bottom, it will support your free hanging features better.

The sort of adaptive infill you propose is definitely something we'd like to have at some point.

Taiko1976, 2017-01-03:
Hi,

thank you for your reply and your consideration. As soon as you have
version 2.4 in beta state I'd like to test the features you've been talking
about. I'll gladly give you some feedback then. Let me know when you are
ready.

Best regards,
Bernd

thopiekar, 2017-01-03:
@Taiko1976, Cura 2.4 is already in beta. Just visit our download page, go to the list of all downloads and scroll down the page.

Regards

Taiko1976, 2017-01-03:
Oh ... great. I'll try to download and test it next weekend. I'll let you
know how it went ;) Thanks again!

Best regards,
Bernd

:

thopiekar, 2017-01-03:
No problem. Feedback is always welcome.

Please don't forget to make a backup in case you are using a set of custom profiles or materials.
Just to be sure, you know :wink:

Taiko1976, 2017-01-03:
I will! Thank you!

:

BagelOrb, 2017-01-03:
Your proposed solution is not likely to be implemented soon. It's quite difficult to efficiently compute where such chambers should be filled.

Do note that the 2.4 is currently in beta; make a backup of your current configuration folder and be aware there might be bugs.

In Cura 2.3 there is also gradual infill, which is more likely to be what you want. Gradual infill decreases from dense to sparse from top surfaces downward. Just set the gradual infill steps to something like 4 and the gradual infill step height to something like 1mm.

Gradual infill is meant to minimise material while maintaining good top surface quality, while cubic subdivision infill is meant for minimising material while maintaining unidirectional strength. Hollow object is an experimental feature, which doesn't work in an optimal way.

closed 2017-01-06 by BagelOrb