Ultimaker/Cura #1285: Cura 2.4 beta Cubic Subdivision infill type not working correctly

rexipus, 2016-12-31:
I just uninstalled Cura 2.3 and installed 2.4 Beta in order to use the Cubic Subdivision infill type. 

When selected, and then the model viewed in layer mode, it is clear that the infill was not generated as expected. Any other infill type will generate the expected infill code, but Cubic Subdivision generates nothing.

This was tested with another, unrelated model, but the infill is also not generated with Cubic Subdivision selected.

Machine is a Windows 10 64-bit installed with recent NVidia drivers installed for a Geforce 1080.

Here is a screenshot showing Cubic  Subdivision infill selected and no infill save for a single line in the upper left arm of this model (hard to see because the rotate arrows partially obscure it):

http://imagizer.imageshack.us/v2/1600x1200q90/924/VJzH5t.jpg

Here is a screenshot of the exact same thing except instead of Cubic Subdivision, just normal Cubic is selected, and the infill is generated as expected.

http://imagizer.imageshack.us/v2/1600x1200q90/921/2MRkAv.jpg

Loki44, 2017-01-01:
I'm having exactly the same issue.

Ghostkeeper, 2017-01-02:
I can't reproduce this myself but I've seen this happen at the installation of @jackha too.

BagelOrb, 2017-01-06:
I have reproduced this as well on my Windows PC

nallath, 2017-01-09:
Is the issue logged on our internal tracker? If so, could anyone post the number here for future reference?

BagelOrb, 2017-01-13:
This issue is a duplicate of https://github.com/Ultimaker/CuraEngine/issues/432

closed 2017-01-13 by BagelOrb