Ultimaker/Cura #1286: 2.3.1 / 2.4 beta macOS material profile does not persist

krunts, 2017-01-01:
Creating and managing materials does not persist when Cura restarts.

*.xml.fdm_material is found under ~/Library/Application Support/cura/materials/ but does not appear in UI.

Importing/exporting of materials also works but does not appear under Materials menu.

[cura.log.txt](https://github.com/Ultimaker/Cura/files/679926/cura.log.txt)

Ghostkeeper, 2017-01-02:
(Under `File`->`Save Project`.)

arodriguezf, 2017-01-02:
I have the same problem in windows 10. Also today all the configuration about my machine had desapeared and Cura starts like the first time you start it, when you have just installed it.

krunts, 2017-01-02:
I just tried it out and this is also happening under Windows 10

Here is the project file:
[projectfile.curaproject.3mf.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/681481/projectfile.curaproject.3mf.curaproject.3mf.txt)

Ghostkeeper, 2017-01-03:
Possibly unrelated bug, but loading that project file gives an error for me:

    2017-01-03 10:37:53,429 - DEBUG - 3MFReader.ThreeMFWorkspaceReader.read [234]: Workspace loading is checking definitions...
    2017-01-03 10:37:53,430 - DEBUG - 3MFReader.ThreeMFWorkspaceReader.read [246]: Workspace loading is checking materials...
    2017-01-03 10:37:53,442 - WARNING - UM.Settings.InstanceContainer.addMetaDataEntry [172]: Meta data with key compatible was already added.
    2017-01-03 10:37:53,596 - DEBUG - 3MFReader.ThreeMFWorkspaceReader.read [274]: Workspace loading is checking instance containers...
    2017-01-03 10:37:53,620 - DEBUG - 3MFReader.ThreeMFWorkspaceReader.read [334]: Workspace loading is checking stacks containers...
    2017-01-03 10:37:53,622 - WARNING - UM.Logger.logException [65]: Exception: We failed to serialize the stack. Trying to clean up.
    2017-01-03 10:37:53,631 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
    2017-01-03 10:37:53,655 - WARNING - UM.Logger.logException [69]:   File "/home/ruben/Projects/Cura/cura/../plugins/3MFReader/ThreeMFWorkspaceReader.py", line 380, in read
    2017-01-03 10:37:53,666 - WARNING - UM.Logger.logException [69]:     stack.deserialize(archive.open(container_stack_file).read().decode("utf-8"))
    2017-01-03 10:37:53,667 - WARNING - UM.Logger.logException [69]:   File "/home/ruben/Projects/Uranium/UM/Settings/ContainerStack.py", line 276, in deserialize
    2017-01-03 10:37:53,667 - WARNING - UM.Logger.logException [69]:     raise Exception("When trying to deserialize %s, we received an unknown ID (%s) for container" % (self._id, container_id))
    2017-01-03 10:37:53,667 - WARNING - UM.Logger.logException [69]: Exception: When trying to deserialize Krunt's Kossel, we received an unknown ID (Krunt's Kossel_settings) for container

lolorc, 2017-01-03:
reproduced on a freshly reinstalled from scratched cura (build from today's master components)
go to material, clone PLA, edit the clone, change name, brand, material (to PETG), color, diameter,
save projet. exit cura
start cura, the newly created material is gone.
easy to reproduce
looks like the created material is save in the wrong file, ".local/share/cura/materials/generic_pla+%232.xml.fdm_material" (i've changed the brand+material fields)

if I clone ABS, just change the the name and color it creates a .local/share/cura/materials/generic_abs+%232.xml.fdm_material but still fails to see it upon restart.

MattWhelan, 2017-01-04:
I just filed issue #1297, which is related to string escaping in the .cfg files.

On disk, the filenames are URL-encoded, but in the .cfg files, they're not encoded.  I wonder if the the pound sign in "generic pla #2" is being interpreted as beginning a line comment (# is the line comment character in INI files).  Extra bad because Cura seems to be generating the offending name...

Jarek-SRT, 2017-01-10:
Linux version has the same issue, so this is not a MacOS specific problem. Seems like all platforms are affected.

gregs007, 2017-01-12:
Also confirmed on MacOS 10.11.6 cura 2.3.1

ChrisTerBeke, 2017-10-09:
Does this still happen in newer versions of Cura?

closed 2018-02-21 by ianpaschal