Ultimaker/Cura #1287: Possible Improvement: Honeycomb infill

TheLinuxNerd, 2017-01-01:
I think it would be nice to see a honeycomb infill option for the infill pattern. I just find honeycomb to be much stronger for some composite prints. 

https://www.illustrativemathematics.org/content-standards/tasks/1126

Ghostkeeper, 2017-01-01:
Did you find that it was stronger in FDM printing?

Honeycomb came under discussion about a year ago too, but it was then predicted it could be weaker due to the retractions involved (or double lines) and the kinks in the infill lines.

TheLinuxNerd, 2017-01-02:
it is stronger but not in ever circumstance. Honeycomb has the strongest ratio of material to strength of any pattern. But that doesn't mean it is the strongest; Triangles would be the best. But I'm pretty cheap with my material, so having this available as an infill pattern option would be great. Thank you for you help :)

nallath, 2017-01-02:
I think cubuic subdivision would even be better. We do have that.

DDDirk, 2017-01-15:
The fact that a perfect honeycomb structure is strong doesn't mean a printed one is. That depends on how you print it.
For example, why use retracts in infill? Oozing is not a problem there, as long as the nozzle stays on the inside (no travel moves over the outer parts). Of course that might lead to underextrusion at the start of the next print move, but that can be solved by priming a bit just before that starts, on the inside of the model.

In my experience underextrusion is the biggest enemy of strength, and there always is some underextrusion, which gest worse with higher flow rates (unless you compesate for that). So I sometimes use nominal overextrusion (up to twice the nominal value, so a 200% nominal extrusion rate). Of course this can cause actual overextrusion (though not as much as you might think), and that would not be desirable for the shell, but for infill it doesn't matter. This might cause a buildup of pressure, so before starting on the shell it might be advisable to let that ooze out, again on the inside of the model, where it doesn't matter (and then maybe prime a bit).
Of course this adds to the printing time, and how much difference that makes depends on the size of the model (and the infill percentage). There are always trade-offs.

But back to the honeycomb structure.
That is really just a practical way to make circles, which are the strongest shape. But circles don't fit together as well as hexagons, leaving open spaces. An actual honeycomb (made by bees) really consists of circles with those spaces filled in, which is a waste of material. But here a printer has an advantage because it can just follow those circle-lines. A disadvantage is that it has to partly move over roads already layed, but because all moves are circular the head can move much faster, and my guess is that that will more than compensate.
Just try this:
Draw a circle in clockwise direction. Upon completion continue to draw an adjoining circle in couterclockwise direction. Upon completion of that you follow one of those two circles 1/6 of the way around and draw a circle that touches both previous circles. Which circle you chose to follow determines in which direction you can continue. If you followed the first circle (and went counterclockwise) you can do the same again, for a total of 6 times, completely encircling the first circle. Now you have to follow the roads already layed for 2/6 of a circle and then choose which circle to follow next, 1/6 of the way around. And then make 12 circles encircling the 7 already drawn. Etc, moving ever further outward.
But you can also choose other directions in which to follow lines already drawn, making different shapes. Or you might start with very big circles and then draw smaller circles inside that. Or come up with other ideas (maybe not just circles). Hours of fun. :)
For more inspiration you might look here: https://en.wikipedia.org/wiki/Circle_packing

Ghostkeeper, 2017-01-22:
An advantage of honeycomb over circles in FDM printing is that two adjoining honeycombs have a greater contact area. This makes the bond stronger and ultimately the entire structure.

Ghostkeeper, 2017-02-18:
The new product manager for Cura at Ultimaker rejected this issue for being too unimportant. This means that the main developers can't spend time on it.

yelneerg, 2017-09-16:
That's too bad. I really like the honeycomb infill the makerbot software does, but I like most everything else better in Cura. How makerbot does it the infill lines never overlap which I find works better than the cura infill options.

DDDirk, 2017-09-17:
Hoow does it work better? Overlapping roads cause more bonding between them, which is likely to make the print stronger. (Although also more stiff, which might be a disadvantage for strength, depending on what you want.)

Ghostkeeper, 2017-09-24:
On some materials, the overlapping lines cause a blob where they overlap and then a gap after the overlap, making the infill weaker.

emphasize, 2017-12-31:
Slic3r has this pattern, so it's available and open source. So where's the big workload to port it to cura? Looking at this (majorly) from a speed perspective doesnt't really reflect the full scope/field of application. Not to mention that the "if retraction-argument" is the weakest possible in light of the complexity of 3dprinting and getting things right.

fieldOfView, 2017-12-31:
> Slic3r has this pattern, so it's available and open source. So where's the big workload to port it to cura?

Saying that is about as productive as a programmer saying "Cura is open source, so why don't you implement it yourself?".

The fact that both Slic3r and Cura are open source is fairly irrelevant here, because of how different the codebases are. I would guess that implementing a honeycomb pattern from scratch is less work than trying to copy and adapt the Slic3r code.

The problem is that there are a lot of bugs to fix and features requested for Cura, and much of that work is seen as more important than adding honeycomb. So much to do, and only so little time.

jackha, 2018-01-02:
One of the drawbacks and reasons that we do not have priority for honeycomb is that it has a lot of corners, so the print also takes a bit longer because of all the (de)accelerations. I would prefer cubic subdivision, but I didn't do a strength test yet.

emphasize, 2018-01-02:
I have found a pattern that is somewhat similar to honeycomb: the tri-hexagon. 

That said im experimenting right now with cubic subdivision in combination with model merging and by setting a minimum infilll area. That gives me great flexibility to decide where to put rigid "subdivisions" (eg 100% infill)

maccesch, 2018-01-05:
I'd like to have honeycomb because in transparent parts it just looks awesome :) Many parts I print I  don't care too much about strength more about looks.

ruben0909, 2018-05-18:
+1

Two links:

http://www.slate.com/articles/health_and_science/science/2015/07/hexagons_are_the_most_scientifically_efficient_packing_shape_as_bee_honeycomb.html

https://www.liebertpub.com/doi/10.1089/3dp.2015.0036

jackha, 2018-07-26:
We decided not to do this.

closed 2018-07-26 by jackha