Ultimaker/Cura #1289: CURA 2.4 BETA crash on start

landway, 2017-01-02:
I installed a new CURA 2.4. After the launch window will appear - loading plugins; updating configuration and window disappear.

Program not running neither in Task Manager.

I have installed 3 version of CURA 15.04.6; 2.3.0 and 2.4.0.

I have W7 64bit

Where could be the problem? Very much looking forward to it.

Ghostkeeper, 2017-01-02:
Possibly a duplicate of https://github.com/Ultimaker/Cura/issues/1283

Could you post your Cura log for us? The log file is stored in `C:/Users/<landway>/AppData/Local/cura/stderr.log`. Note that AppData is a hidden folder (I often find it easy to just type the address in Explorer).

landway, 2017-01-02:
Possibly a duplicate of #1283 - thanks for tip, but i can not start cura neither to main window...

LOGs:

[stderr.txt](https://github.com/Ultimaker/Cura/files/680673/stderr.txt)
[stdout.txt](https://github.com/Ultimaker/Cura/files/680674/stdout.txt)
[cura.txt](https://github.com/Ultimaker/Cura/files/680675/cura.txt)

I am not sure if this is right logs, because I have 3 version of cura instaled...

Ghostkeeper, 2017-01-02:
This is the error it's breaking on.

    2017-01-02 10:43:44,773 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
    2017-01-02 10:43:44,788 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
    2017-01-02 10:43:44,788 - CRITICAL - cura.CrashHandler.show [35]:   File "C:\Python35\lib\site-packages\cx_Freeze\initscripts\Console.py", line 21, in <module>
    2017-01-02 10:43:44,804 - CRITICAL - cura.CrashHandler.show [35]:   File "D:/2.4/build/inst/bin/cura_app.py", line 61, in <module>
    2017-01-02 10:43:44,804 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\Application.py", line 317, in getInstance
    2017-01-02 10:43:44,804 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 144, in __init__
    2017-01-02 10:43:44,820 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
    2017-01-02 10:43:44,820 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 106, in upgrade
    2017-01-02 10:43:44,820 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 292, in _upgradeFile
    2017-01-02 10:43:44,820 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\MimeTypeDatabase.py", line 156, in getMimeTypeForFile
    2017-01-02 10:43:44,835 - CRITICAL - cura.CrashHandler.show [35]: UM.MimeTypeDatabase.MimeTypeNotFoundError: Could not find a valid mime type for .\tmpuhe3lvzx

Could this be the lock file that's created during the loading process, @jackha ?

glovax, 2017-01-02:
Hallo to all. And this is mine :-):
`2017-01-02 16:47:19,114 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ConsoleLogger
2017-01-02 16:47:19,610 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: f:\Program Files\Cura 2.4\CuraEngine.exe
2017-01-02 16:47:19,611 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraEngineBackend
2017-01-02 16:47:19,627 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileReader
2017-01-02 16:47:19,660 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileWriter
2017-01-02 16:47:20,129 - INFO - UM.PluginRegistry.loadPlugin [83]: Plugin Doodle3D uses an incompatible API version, ignoring
2017-01-02 16:47:20,334 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJReader
2017-01-02 16:47:20,367 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJWriter
2017-01-02 16:47:20,384 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLReader
2017-01-02 16:47:20,417 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLWriter
2017-01-02 16:47:20,447 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-01-02 16:47:20,479 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-01-02 16:47:20,499 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2017-01-02 16:47:20,680 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-01-02 16:47:20,808 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2017-01-02 16:47:20,896 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-01-02 16:47:20,928 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-01-02 16:47:21,104 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2017-01-02 16:47:21,481 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-01-02 16:47:21,483 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-01-02 16:47:21,595 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PostProcessingPlugin
2017-01-02 16:47:21,809 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-01-02 16:47:21,859 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-01-02 16:47:21,937 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-01-02 16:47:22,039 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-01-02 16:47:22,709 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-01-02 16:47:22,900 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-01-02 16:47:22,966 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-01-02 16:47:23,052 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-01-02 16:47:23,470 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-01-02 16:47:23,838 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2017-01-02 16:47:24,098 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-01-02 16:47:24,178 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UpdateChecker
2017-01-02 16:47:24,179 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-01-02 16:47:24,670 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2017-01-02 16:47:24,814 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2017-01-02 16:47:24,847 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2017-01-02 16:47:24,877 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-01-02 16:47:24,918 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-01-02 16:47:25,074 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2017-01-02 16:47:25,115 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-01-02 16:47:25,165 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-01-02 16:47:25,167 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'external-backend': False, 'file': [], 'disable-textures': False}
2017-01-02 16:47:25,186 - INFO - UM.VersionUpgradeManager.upgrade [99]: Looking for old configuration files to upgrade.
2017-01-02 16:47:25,199 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: 'NoneType' object is not iterable
2017-01-02 16:47:25,201 - WARNING - UM.Logger.logException [69]: Traceback (most recent call last):
2017-01-02 16:47:25,201 - WARNING - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 305, in _upgradeFile
2017-01-02 16:47:25,203 - WARNING - UM.Logger.logException [69]: TypeError: 'NoneType' object is not iterable
2017-01-02 16:47:25,241 - INFO - UM.VersionUpgradeManager._upgradeFile [343]: Upgraded .\K8200+%233.stack.cfg to version 3.
2017-01-02 16:47:25,316 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
2017-01-02 16:47:25,318 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
[... cut ...]
2017-01-02 16:47:25,330 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 292, in _upgradeFile
2017-01-02 16:47:25,332 - CRITICAL - cura.CrashHandler.show [35]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\MimeTypeDatabase.py", line 156, in getMimeTypeForFile
2017-01-02 16:47:25,333 - CRITICAL - cura.CrashHandler.show [35]: UM.MimeTypeDatabase.MimeTypeNotFoundError: Could not find a valid mime type for .\Vellman+K8200.curaprofile
`

Ghostkeeper, 2017-01-03:
Ah, glovax's case is indeed the same as #1283, not the same as what landway is seeing.

landway, 2017-01-03:
Partially solved by uninstalling all the older versions, deleting all files and folders that are associated with CURA, than clean install a new version ... Thank you for your effort.

lolorc, 2017-01-03:
"partially solved" by removing all your settings !!??? :) you're too nice ;-)

landway, 2017-01-03:
What do you mean? Left me nothing. We need test a new CURA. Do you have some idea? Or I missed something?

lolorc, 2017-01-03:
just playing with words. there's a nasty bug which needs to be fixed. We can't help in testing a development tree if we had to start from scratch every day. Or If we have to, we need some way to manually re-import settings.

glovax, 2017-01-03:
@Ghostkeeper , sorry that i mismached topics. I fought errors similar, but realy do not know where error is in fact.

landway, 2017-01-03:
Maybe simpli a lot of versions on the heap :)

Ghostkeeper, 2017-01-04:
@glovax That's okay. It appears the same for the user but the error message is different. It's my expertise that can differentiate ;)

The only configuration files that Cura ever creates are in `C:\Users\<username>\AppData\Local\cura`, so deleting that folder should always make Cura behave like a fresh install. Since this is a bug with the configuration upgrade system, deleting all that configuration is a workaround for now.

I filed this as a blocking bug for the 2.4 release. If my boss agrees that it's blocking, it must get fixed for 2.4.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-10:
Fixed in 2.4.