Ultimaker/Cura #1291: Support Z distance

maukcc, 2017-01-03:
Cura 2.4.0
OS ubuntu 16.04

- There is always a layer of empty space between the support and the object
- The support Z distance is rounded of to a full layerheight

So if you have a layerheight of 0.2mm and support Z distance of 0.1mm, there is 0.4mm of space between support and object.(this is the case with and without an interface layer)

Ghostkeeper, 2017-01-04:
The rounding is intentional. We need to have the layers of the support aligned with the other layers.

There shouldn't always be an empty layer between support and object. Not if the Support Z Distance is 0 and there is no mesh next to it to influence this via the Support X/Y distance.

maukcc, 2017-01-05:
For now you  could rename the "support Z distance" to "support Z distance layers" or put some info about this in the help text pop-up.
In the end you really need your support to end (or start) at a different layerheight as the layers of the object. Especially in the future when you will be using different layerheights for support ;)

BagelOrb, 2017-01-06:
I moved it (back) to 2.5, but why it was moved from 2.4 to 2.5 is still a mystery.

s-imphead, 2017-03-03:
Thanks to maukcc for well-chosen example! I've been completely perplexed by this bug.  The description is found in [12 source files](https://github.com/Ultimaker/Cura/search?utf8=%E2%9C%93&q=%22provides+clearance%22&type=Code), but I'm not sure how build system works (and not sure what CURA-2364 refers to) but to get the ball rolling:
```
"Additional distance from the top/bottom of the support structure to the print. The sum of" 
"the Support Z Distance and the layer height is rounded up to nearest multiple of layer height."
"This resulting gap provides clearance to remove the supports after the model is printed. 
```
BagelOrb or GhostKeeper, if you have commit privileges and if this is useful, please turn into commit... Thx

BagelOrb, 2017-03-03:
I don't understand what you mean.

About the 12 source files: one is the actual source file, the rest are translation files for other human languages.

What do you want us to commit?

s-imphead, 2017-03-03:
In the source file (whatever its name is!), I suggest replacing
```
"Distance from the top/bottom of the support structure to the print. This gap "
"provides clearance to remove the supports after the model is printed. This "
"value is rounded down to a multiple of the layer height."
```
with
```
"Additional distance from the top/bottom of the support structure to the print. The sum of" 
"the Support Z Distance and the layer height is rounded up to nearest multiple of layer height."
"This resulting gap provides clearance to remove the supports after the model is printed."
```

That's all -- I presume that text will flow to the pop-up hint for setting the Support Z Distance?

Ghostkeeper, 2017-03-06:
We have since updated the text of our own to say that it rounds up.

I just re-tested it and currently the Support Top Distance setting is correct. At support bottom distance is sometimes incorrect due to the Support Stair Step Height setting. We're in the process of fixing it. The two people opposite of my desk are currently discussing that as I type.

s-imphead, 2017-03-06:
Great, thx.  I'm including an example for reference if others encounter the same problem as I did.   With default settings (High Quality, 0.06mm, 0.2mm nozzle)  I'd get something like this for severe overhangs, where the support is retracted from outer wall in areas of very high curvature:
[image]
Setting Z Support Distance to 0 solves this problem.

BagelOrb, 2017-07-02:
I conclude from this conversation that the problem is solved.

closed 2017-07-02 by BagelOrb