Ultimaker/Cura #1292: Layerheight for support

maukcc, 2017-01-03:
A separate  layerheight for support material would be usefull to speed up the process.

Ghostkeeper, 2017-01-04:
We need to have the layers of support aligned with the rest of the layers. This is a limitation in our current structure, but you can imagine the nightmare of complexity it would be to plan travel moves between those and avoid printed objects in those travel moves.

We've thought about introducing a similar setting to Combine Infill Layers, where it would combine for instance 3 layers of support together, removing support from two of those layers but printing the third with thrice the flow. But this could seriously compromise the stability of the support while it's printed.

DDDirk, 2017-01-04:
If the objective is to print faster you can simply print the support at a higher speed (and thus a higher material flow). However, if you are already pushing the limits for the print itself that might be a bad idea. But if you print at high quality (low material flow) that would be a very simple solution. The support doesn't have to be pretty, it just has to be there.
I am not sure, but I believe that given the flow rate a higher print speed gives better results than higher layerheight. Within limits of course.
Mind you, after a change in material flow results can be unpredictable, so it would be best not to print the outer wall after having printed support at a higher speed.

A more complicated option would be to use increasing support density, as with Recursively Sub-divided Cubic Infill.

maukcc, 2017-01-05:
Yes you could print faster, but printing with higher speeds will only make a marginal difference in time and will deteriorate the quality/toughness of the object.  
Printing infill and or support with a bigger nozzle and therefore bigger width, lets you also print at bigger layerheights. This results in a dramatic decrease of time to print an object.

DDDirk, 2017-01-06:
How much printing speeds up with faster printing of the support depends a lot on the model. But of course it is one of the things that could help, so why not implement it?
But how would it affect the quality of the print? All support has to do is, well, support the model. The quality of the support might have to be better where it touches the model, so maybe the speed could be lowered there.

maukcc, 2017-02-10:
It is not just the layerheight for support that I am interested in. I would like separate layerheight settings for:
- shell
- support
- support interface
- infill
This will reduce print time to 20% (not to be mistaken for: reduce print time with 20%)

Ghostkeeper, 2017-02-13:
Add the layer height (or rather, layer-combine-count) of walls as well. Then you just take the actual layer height to be the greatest common divisor of all those layer heights and you'll end up with actual fully adjustable layer heights for every part. Though the greatest common divisor might end up to be 1µm which takes a long time to slice.

I think we'd all agree that this is desirable, or at least the end goal of fully adjustable layer heights for each part. But it takes a lot of work to make this happen!

DDDirk, 2017-02-15:
This might also make prints stronger because there is no clean-cut separation across the print between layers.
I've been playing with this idea a few months ago, but haven't gotten around to testing it yet.

maukcc, 2017-04-20:
We have just tested the "Infill Layer Thickness", and that works like a charm.
Is there any way to quickly copy that function to the support?

Ghostkeeper, 2017-04-24:
No, but this is one of the features we want to add with quite high priority. It's currently not possible.

maukcc, 2017-07-13:
Whooaa, I see a "Support infill layer thickness" value.
fantastic !
But it does not seem to do anything. Is this still a work in progress, or am I doing something wrong?

Ghostkeeper, 2017-07-19:
You might be. It works here.

What I did was:
* Load a model that requires support.
* Enable support.
* Set Infill Layer Thickness to 0.5mm.
* Switch to Layer View.

You see thicker lines for support (5 times as wide) but these lines are only seen every 5 layers. That means it's working.

Ghostkeeper, 2017-07-19:
Ah, by the way, it doesn't work on the wall of support, so if you have Grid Infill or something on your support, only the lines on the inside will be thicker.

Ghostkeeper, 2017-08-07:
Closing this as it has been implemented now and fairly thoroughly tested.

closed 2017-08-07 by Ghostkeeper

maukcc, 2017-08-15:
It does not work here.
Yes, it works on infill.
No, it does not work on support

maukcc, 2017-08-16:
OK that is strange.
It works in 2.7 appimage, but not in 2.6.99 master