Ultimaker/Cura #1293: Cura 2.3.1 Crashing

NatAllen22, 2017-01-03:
Hi, I have recently installed Cura version 2.3.1 but it keeps crashing shortly after importing a model.
Application Version: Cura 2.3.1
Platform: Windows 7 Enterprise
Processor: Intel Xeon CPU E3-1505M v5 @ 2.80GHz
RAM: 32.0GB
System Type: 64 - bit Operating System

I have uploaded the cura log and dxdiag in .docx format 

[Cura Log.docx](https://github.com/Ultimaker/Cura/files/682069/Cura.Log.docx)
[dxdiag.docx](https://github.com/Ultimaker/Cura/files/682078/dxdiag.docx)

Cheers, Nat

NatAllen22, 2017-01-03:
Here is also the Problem signature from when it crashes
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	Cura.exe
  Application Version:	0.0.0.0
  Application Timestamp:	54467a5f
  Fault Module Name:	ig9icd64.dll
  Fault Module Version:	20.19.15.4444
  Fault Module Timestamp:	571a907d
  Exception Code:	c0000005
  Exception Offset:	0000000000586e9d
  OS Version:	6.1.7601.2.1.0.256.4
  Locale ID:	2057
  Additional Information 1:	6601
  Additional Information 2:	6601502705a303aa2714b4a42a8776ba
  Additional Information 3:	e673
  Additional Information 4:	e673c8bcb3881ed21d4d77b2bca7c6a7

thopiekar, 2017-01-03:
Googled about the module (*.dll file) which is mentioned in the problem signature.
The file seems to be part of the Intel Graphics driver. Could you be so kind and double check, whether your driver is up-to-date?

https://downloadcenter.intel.com/

Thanks :wink:

NatAllen22, 2017-01-03:
Hi, thanks for your quick response!
I have an Intel HD Graphics P530 and the driver version is 20.19.15.4444 is this the latest version?
I am not able to download the Intel driver update utility due to not having permission on the corporate laptop :(

thopiekar, 2017-01-03:
Hmm, probably you also won't be able to update the driver then.

However, this is the latest driver, which mentions you GPU:
https://downloadcenter.intel.com/download/26464/Intel-Grafik-Beta-Treiber-15-45-?wapkw=intel+hd+graphics+p530

nallath, 2017-07-27:
I'm closing this as it doesn't seem to be an issue with Cura (but with your local drivers)

closed 2017-07-27 by nallath