Ultimaker/Cura #1294: Opening project file NOT the same as loading a project file (tested on OSX)

filipgoc, 2017-01-03:
Maybe this is desired behavior, but it feels confusing: 

## Expected, good
When opening a .curaproject.3mf file from Cura (File > Open Project)
- .stl configuration on the build plate loaded
- build plate cleared
- print settings loaded

## Unexpected, bad?
When opening a .curaproject.3mf file from a file browser, 
(effectively adding stls, but doing nothing else)
- .stl configuration on the build plate loaded
- build plate NOT cleared 
- print settings NOT loaded
- NO open project dialog

Ghostkeeper, 2017-01-04:
Ah, hmm, I think it then interprets the file as if it were a normal .3MF file, by the looks of things.

filipgoc, 2017-01-04:
Makes sense.

nallath, 2017-07-27:
This has been fixed.

closed 2017-07-27 by nallath