Ultimaker/Cura #1297: Custom profile names containing comma break machine

MattWhelan, 2017-01-04:
1 Create a new custom profile named "Normal, with support", based off of the "Normal" profile, but with supports enabled.
   2 Exit Cura
   3 Start Cura

The first-time startup wizard will run, because the comma in the profile name isn't escaped in the Stack cfg file, and the file uses CSV-encoding.

I suggest disallowing commas in custom profile names, since you seem to be pretty stuck with CSV-encoding stuff in your INI file values.  Long-term, maybe switch to YAML or JSON.

Ghostkeeper, 2017-01-04:
We should've fixed this in Cura 2.4. We changed the layout a bit of the .cfg files.

Were you using 2.3.1?

MattWhelan, 2017-01-04:
Yes, I was using 2.3.1. I also tried 2.3.0.

closed 2017-01-05 by nallath