Ultimaker/Cura #1299: Cura 2.4.0 PPA crash on startup on Ubuntu

skorokithakis, 2017-01-04:
Cura crashes at startup, even through a reinstall and without any settings. Here's my log:

~~~
2017-01-04 14:39:01,709 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ConsoleLogger
2017-01-04 14:39:01,716 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-01-04 14:39:01,716 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraEngineBackend
2017-01-04 14:39:01,718 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-01-04 14:39:01,719 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-01-04 14:39:01,722 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-01-04 14:39:01,727 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-01-04 14:39:01,728 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-01-04 14:39:01,731 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-01-04 14:39:01,732 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-01-04 14:39:01,733 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-01-04 14:39:01,734 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-01-04 14:39:01,736 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-01-04 14:39:01,736 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin ConsoleLogger was already loaded
2017-01-04 14:39:01,737 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLWriter
2017-01-04 14:39:01,740 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-01-04 14:39:01,741 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin STLReader
2017-01-04 14:39:01,742 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJReader
2017-01-04 14:39:01,743 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin OBJWriter
2017-01-04 14:39:01,744 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-01-04 14:39:01,746 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-01-04 14:39:01,748 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2017-01-04 14:39:01,753 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2017-01-04 14:39:01,758 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFReader
2017-01-04 14:39:01,760 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-01-04 14:39:01,779 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-01-04 14:39:01,781 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin 3MFWriter
2017-01-04 14:39:01,784 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-01-04 14:39:01,786 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileWriter
2017-01-04 14:39:01,787 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2017-01-04 14:39:01,789 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2017-01-04 14:39:01,791 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2017-01-04 14:39:01,792 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2017-01-04 14:39:01,794 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ChangeLogPlugin
2017-01-04 14:39:01,795 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin AutoSave
2017-01-04 14:39:01,803 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2017-01-04 14:39:01,804 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-01-04 14:39:01,810 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2017-01-04 14:39:01,818 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-01-04 14:39:01,821 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-01-04 14:39:01,822 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CuraProfileReader
2017-01-04 14:39:01,823 - WARNING - UM.PluginRegistry.loadPlugin [72]: Plugin CuraEngineBackend was already loaded
2017-01-04 14:39:01,882 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-01-04 14:39:01,886 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-01-04 14:39:01,886 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'disable-textures': False, 'external-backend': False, 'file': []}
2017-01-04 14:39:01,911 - INFO - UM.VersionUpgradeManager.upgrade [99]: Looking for old configuration files to upgrade.
2017-01-04 14:39:01,913 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-01-04 14:39:01,918 - INFO - UM.Backend.Backend.startEngine [71]: Started engine process: /usr/bin/CuraEngine
2017-01-04 14:39:01,918 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '']
2017-01-04 14:39:01,929 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-04 14:39:01,929 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2017-01-04 14:39:01,929 - DEBUG - UM.Backend.Backend._logSocketState [171]: Socket state changed to Connected
2017-01-04 14:39:01,929 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2017-01-04 14:39:01,930 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-04 14:39:01,930 - DEBUG - UM.Backend.Backend._onSocketStateChanged [161]: Backend connected on port 49674
2017-01-04 14:39:01,930 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
2017-01-04 14:39:01,930 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-01-04 14:39:01,933 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-01-04 14:39:01,933 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] (at your option) any later version.
2017-01-04 14:39:01,934 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-04 14:39:01,934 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is distributed in the hope that it will be useful,
2017-01-04 14:39:01,934 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-01-04 14:39:01,934 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-01-04 14:39:01,934 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] GNU Affero General Public License for more details.
2017-01-04 14:39:01,935 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-01-04 14:39:01,935 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-01-04 14:39:01,935 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-01-04 14:39:01,972 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3; Multi-extrusion printers are not supported
2017-01-04 14:39:01,993 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ord; Multi-extrusion printers are not supported
2017-01-04 14:39:02,055 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
2017-01-04 14:39:02,465 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to punchtec_connect_xl; Multi-extrusion printers are not supported
2017-01-04 14:39:03,205 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3_extended; Multi-extrusion printers are not supported
[... cut ...]
2017-01-04 14:39:08,782 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2017-01-04 14:39:08,839 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [175]: Could not get metadata of container  because it was not found.
fish: “cura” terminated by signal SIGSEGV (Address boundary error)
~~~

skorokithakis, 2017-01-05:
I am, yes.

thopiekar, 2017-01-05:
Then this is a known problem. The only solution is to choose a different desktop then Unity7 or Unity8.
The problem has already been reported here and finally to Canonical, see #1094.
Additionally there is a link in #1094 to the bug report at launchpad.net. If you've got a account at launchpad.net please mark yourself there that this problem affects you. The more people are affected by a issue on launchpad.net, the more attention it will get by the maintainers (here Canonical, as Unity is their desktop environment).

Thanks!