Ultimaker/Cura #1300: "Jumping" in wall (2.4 beta)

Ando120, 2017-01-04:
Example: if you create and want to print solid wall 1.5 mm and rotate it 30 degrees then every second layer middel print line is "jumping".
Its visible in cura layer view.

That bug appear differently when wall is in different angle(z). 
Also it looks and sound badly when printing big circle wall.

Ghostkeeper, 2017-01-05:
I can't reproduce this.

I loaded a cube. Then I resized it to x: 1.5mm, y: 20mm and z: 20mm. Rotated it 30 degrees around the Y axis (green circle). Every layer is exactly the same, except shifted a bit in the X direction.

What did you mean exactly by "print solid wall 1.5mm and rotate by 30 degrees"? And by "jumping"? Do the lines move around?

Ando120, 2017-01-05:
rotate it around Z axis.

Ghostkeeper, 2017-01-05:
Ah, like this.

[image]

That would indeed be undesirable. It's the result of the print small gaps feature, I think. @BagelOrb certainly has more to say about that, but he's lots behind on reading his e-mail at the moment.

BagelOrb, 2017-01-06:
This is the same underlying problem as https://github.com/Ultimaker/Cura/issues/1303

jackha, 2018-07-30:
Deferred because of inactivity

closed 2018-11-26 by Appesteijn