Ultimaker/Cura #1302: Cant select any roof settings in Cura

ArtSosa, 2017-01-04:
My problem is basically the title its self i cant select any roof settings in Cura i do not know what i'm doing wrong please help if possible. I'm sure its a silly mistake of user error. Sorry for the inconvenience 

Best regards,
 Sosa

 
[image]

fieldOfView, 2017-01-04:
You see those "i" icons? Try hovering your mouse over them

ArtSosa, 2017-01-04:
It say "this setting has been hidden by the values of enable support interface, enable support, support interface extruder, Change the values of those settings to make this setting visable." i have tried but it still does not activate or remove the "i" icon from this setting.

fieldOfView, 2017-01-04:
Realise that the setting visibilty dialog makes settings visible in the sidebar, it does not change those setting values. "Support Interface" is not visible in your sidebar. If you do make it visible, you will see that its value *in the sidebar* is unchecked. Tick the checkbox in the sidebar, and the roof settings will become visible.

ArtSosa, 2017-01-05:
Hey it worked thank you so much. =)

closed 2017-01-05 by nallath

nickthetait, 2017-01-05:
Here's an idea on how we can improve the labeling for this message:
"This setting cannot be displayed until the following settings are visible and enabled: {setting names}."

ArtSosa, 2017-01-05:
For real @nickthetait  i dont like how that is

Ghostkeeper, 2017-01-06:
That wouldn't work well for settings that are disabled because other settings are below/above some numeric threshold or because of an enumerated setting, such as settings that are only for multi-extrusion, or on infill density being greater than 0, or dependent on bed adhesion being set to "Raft".