Ultimaker/Cura #1303: Cura 2.3.1 - walls empty inside even though 100% filling chosen/

dimbmw, 2017-01-04:
Hi
I have Ultimaker 3 and experiencing issues with the software.
I fill 100% and still see that there are empty spaces inside the walls.

The only work around for now is to set "wall line count" parameter to 0. Then the inner space fills completely. But this is not a solution when you set, say, 50% fill, as then you only get a fill structure, with no outer walls!

Am I missing something or is it a bug ?  I could not get any help from the US support, they are using older versions and they tried my STL and it fills OK with the older virsions, the problem is with the version 2.3.1. 
But this is the only one I have.

[image]

Ando120, 2017-01-05:
Same was for me, but its fixed in 2.4 BETA

dimbmw, 2017-01-05:
Actually the bug improved a little bit in the ver 2.4 beta, but it is still there - see the screenshot.

[image]

Ghostkeeper, 2017-01-05:
Indeed. This is one that was reported often during the lifetime of 2.3. We've fixed it in the meanwhile.

dimbmw, 2017-01-05:
It is not fixed in ver 2.4 beta, the problem still persists.

Ghostkeeper, 2017-01-06:
Ah, I see now that you said it improved but it's still there.

dimbmw, 2017-01-06:
Yes, the problem is still there.
Check this for example - the white arrows are showing hollow volumes, while it should be 100% filled.

[image]

BagelOrb, 2017-01-06:
The way in which the gap filling works makes that it doesn't work that good for diagonal gaps.
Like the infill, the gaps are filled with diagonal lines, which then get merged into lines which follow the contour of the gaps.

However, for small gaps which are aligned with the infill direction, the diagonal lines may just skip the hole and no infill will be generated.

If you rotate your part 45 degrees more gaps will be filled in this particular print.

Fixing this bug is going to take quite some work, so don't expect it to work flawlessly soon.

I still want to look into this and see if we may have made another mistake when filling gaps between walls.

Ando120, 2017-01-06:
If I create a big cylinder then most part it print along circle but two areas about 45 degrees it is printing like a infill pattern. 
So, it looks as bug.
If you can tell where "gap filling" part of code is  (file and start line )  maybe I can find the bug.

BagelOrb, 2017-01-07:
That's not a bug in that sense. It's a integral part of the way in which
infill lines are combined into lines which follow the gap. When the lines
are too far apart, thy cannot be combined because the combining could then
sometimes combine infill lines of totally different parts.

If I recall correctly then the code to combine infill lines into gap
following lines is in a separate class in MergeInfillLines.cpp. I'm
currently not at my PC, so I'm not sure. It's called somewhere during
gcodePlanner::writeGcode(.)

dimbmw, 2017-01-08:
In the example above, clearly the software was supposed to do concentric filling (and actually the concentric infill was chosen in settings), so it is weird that all of a sudden it stopped feeding the material. 
I don't find these areas any different as compared to others on the same circle.

[image]

BagelOrb, 2017-01-08:
Dont get me wrong - it is a bug.

However it is a natural limitation of the chosen approach for gap filling. To solve this bug we need to rewrite the whole gap filling algorithm.

dimbmw, 2017-01-08:
Same part slices by Simplify3D:

[image]

It is of course a pain that a $3500 printer is using a community supported software and the manufacturer is not responsible for anything about it.  I quote from the email I got from the USA support:  "CURA is an open-sourced software. This means that you are welcome to edit the code as you wish. "  Really, Ultimaker? So now after I paid $3500 you suggest me to go fix the software myself? Seriously? Printing solid filled parts is a basic ability of any basic printer, my older $500 flashforge does it. Oh, sorry, I forgot, i can't blame the printer, it is all community software...

BagelOrb, 2017-01-09:
Dear dlmbmw,

You seem to have misunderstood how the Cura project is set up. Cura is developed by us at Ultimaker, but we keep it open source so that anybody which wants to contribute to the project is able to do so. The software is made specifically for our Ultimaker printers, but by accepting Pull Requests of 3rd parties, the software is also able to support 3rd party printers.

Of course you are welcome to edit code, but that should never be used as an argument to say that you should fix some problems you are having yourself. It's not really clear to me whether the USA support team sent you an email which said you should fix the problem yourself, or whether you misinterpreted their support email. Could you maybe provide the name of the support employee with which you have been in contact?

In your screenshot above it seems you have used different line width settings than you have used in Cura. If you tweak you line width settings a bit you might get the area to be filled in by normal walls after all - just like S3D does.

Note that S3D also suffers from the gap filling problem:
[image]

dimbmw, 2017-01-09:
BagelOrb:  I've relpied in the email, did not want to publish names here. 
Thank you for the clarification. I am relieved that I do not have to edit the Cure code myself. I hope you can come up with some solution.

BagelOrb, 2017-01-10:
Your email contains a lot of information, so I'm copying the relevant parts here so that I can respond to them.

> I would really prefer to deal with some more knowledgeable people to make this printer work.

You have come to the right place, my friend ;)

> [M]y PLA is not sticking to glass, the tip-glass distance gap is too wide, I ask a specific question on what distance between the tip of the nozzle and the glass the auto leveling thing sets. 

The gap should be zero, but the first layer should be printed at a distance of `Initial Layer Height` from the bed.
Did you measure the gap between the nozzle and the bed?

There are lost of other tricks to enhance the bed adhesion; you should google what people do for build plate adhesion. I find that keeping it clean works well.

> Clearly the auto leveling thing is a great idea, but it apparently needs to be calibrated, as some manufacturing/assembly tolerances could have bring an extra 0.05 mm that prevent it from level properly. If it had some engineering menu where one could calibrate the auto leveling feature that could really help. But at this time i have to still level the bed manually.

Autoleveling compensates for how slanted your bed is. It's always a good idea to get your bed as leveled as possible, so that the printer doesn't need to compensate that much.

> [H]ow do I tell the machine not to drop the print head into the initial purge (as sometimes it drags the purge on a part of the purge in the print area). [...] [A]ctually it is clear that it is not about editing the cura code, but about editing the initial section of the g-code sent to the printer, so it sounds easy to fix if the initial g-code section is available for editing. 

I think that is a bug which is fixed in Cura 2.4. The way it now works is that a poop is printed after which the nozzle move diagonally downward in order to wipe the poop in the glass. The downward move is diagonal so that the nozzle doesn't dive into the poop. If the nozzle then moved horizontally into the poop then that was a bug in Cura; the diagonal move downward is part of the firmware on the printer - not a part of Cura.

The prime poop is actually not part of the start gcode, because the second material is first used at the middle of the first layer. It then only primes the second extruder after it has printed with the first.

> My biggest problem now is that I can not have the UM3 print any high quality prints.
[image]
[image]
> Take a look at these 2 same parts - one printed on UM3, and another one on a cheap $500 flashforge finder. Look at the details of the circled areas. Flashforge finder prints so much better. Check a couple of pictures attached. I bought the UM3 thinking that my print quality will improve vs flashforge finder, but as of now I am struggling to even make it the same.

It looks as though you are printing on different temperatures on those two pictures.

You mentioned that you are using your own kind of PLA. Currently Cura doesn't support third party materials that well. I advise you simply select the Recommended settings for PLA at normal quality without tweaking any settings. Then tune the temperature on your printer to account for the difference in required temperature which are listed on Ultimaker PLA rolls and your own PLA roll.

Tuning the temperature on the printer applies the temperature tuning to all temperature settings at once. Print temperature is not a simple setting like it used to be; there's initial layer temp, starting temp, final temp, print temp, standby temp and all kinds of settings controlling the heatup and cooldown speed.

partsoven, 2017-01-19:
Can I ask what awesome screenshot app you are using?

dimbmw, 2017-01-19:
It is evernote, you can download it for free.

partsoven, 2017-01-19:
I have Evernote and use it often didn't even know it did this. Ah-hah! Thanks!

dimbmw, 2017-01-19:
Just right click on the evernote icon in the right lower corner, select "copy selection", and clip the desired area... works like a charm.

Ghostkeeper, 2017-01-22:
> It looks as though you are printing on different temperatures on those two pictures.
To me it looks like this would be the result of making tiny movements for those pieces of infill: The blur of travel moves that we see in dimbmw's layer view screenshots.

ianpaschal, 2018-05-22:
@Ghostkeeper This is more than a year old. Do you think some of the changes we've made to walls would eliminate these problems?

BagelOrb, 2018-05-22:
Nope.

Ghostkeeper, 2018-06-29:
The travel moves might be a bit better with the latest changes to the combing collision areas, but the gaps problem is still there.

Vandrasc, 2019-06-24:
@dimbmw is this still the case in the latest 4.1 Cura (walls empty inside even though 100% filling chosen)?

BagelOrb, 2019-06-25:
@Vandrasc I don't think the original poster is still involved in this issue after a couple of years. I think it's a better idea if you test it yourself.

mahtDFR, 2020-02-19:
There has been no activity on this issue for 30 days, so I am closing it.

BagelOrb, 2020-02-20:
This bug should be solved with the new libArachne implementation which uses variable line width to fill in these gaps using wider lines, while reducing the line widths of the walls.

Ghostkeeper, 2022-03-14:
Fixed in Cura 5.0.

closed 2022-03-14 by Ghostkeeper

rtpiil, 2022-04-24:
@Ghostkeeper, I still miss the top layer in version 5.0:
[image]

As can be seen the top most layer is not marked "shell". It is not a serious problem, but still unsatisfying :)

The model is downloaded here: https://www.thingiverse.com/thing:2989026

[image]