Ultimaker/Cura #1304: Improvement Request: Import Multiple Profiles At Once

filipgoc, 2017-01-05:
## Now: 
Only one quality/ material profile can be imported at a time. 
That makes the importing business quite a pain if one has more than one profile to import. 
[image]

## Desired: 
Select as many quality profiles/ material profiles (even a folder maybe? oh my, let's not get too risky) as desired and import all of them. 
[image]

nallath, 2017-01-05:
I've logged this feature in our internal tracker. For our reference; CURA-3207

ianpaschal, 2018-02-27:
This has been marked `Won't Fix` internally.

closed 2018-02-27 by ianpaschal

Ghostkeeper, 2018-03-19:
It was deemed not important enough because it will save some users a few minutes only once per installation.