Ultimaker/Cura #1305: Cura does not migrate settings between versions

skorokithakis, 2017-01-05:
I just installed Cura 2.4.0 beta, and it seems that none of my settings were preserved. Given that I had already set up materials, printers and presets, this is a major inconvenience. Cura should have a migration function that will change my settings to the new format, if necessary.

Ghostkeeper, 2017-01-05:
We have a migration function. It seems to be somewhat broken though. It's on our list of blocking issues for the actual release of 2.4.

skorokithakis, 2017-01-05:
Ah, thank you, if there's a "migration option is broken" ticket, please close this one.

Ghostkeeper, 2017-01-05:
Well, it is the cause of #1289 and #1283, but those don't have exactly the same symptoms as you.

skorokithakis, 2017-01-05:
Hmm, right. Yes, mine was slightly different, although I can't tell if the settings weren't preserved because of a crash...

nallath, 2017-07-27:
I think this has been fixed (as it was a blokcing issue). So this can be closed.

closed 2017-07-27 by nallath