Ultimaker/Cura #1306: Cura sets all temperatures 5 degrees higher.

skorokithakis, 2017-01-05:
I'm using the Cura 2.4.0 beta AppImage, and setting the extruder temp to anything (e.g. 214 in this case), produces this:

~~~
M104 S219
M109 S219
~~~

This happens consistently, Cura always outputs five degrees higher.

ntoff, 2017-01-05:
can confirm, I started a print a few minutes ago and noticed this as well.

Ghostkeeper, 2017-01-05:
That's the print temperature for the initial layer. It defaults to the normal print temperature +5. The setting is called "Print Temperature Initial Layer".

I understand that it's a bit opaque in how this works. What would be a better way to do this sort of thing, you think?

skorokithakis, 2017-01-05:
Oh, sorry, this is fine the way it is, then, I was just confused because I hadn't seen this in 2.3.1. I was trying to see if #1209 was resolved, but it seems that it's not.

I don't think anything should be done, this is a good setting, with an appropriate default and well-named. I was just confused because I didn't know it had been added. I'll close this ticket now.

closed 2017-01-05 by skorokithakis