Ultimaker/Cura #1307: Start gcode isn't at the start

ntoff, 2017-01-05:
**What were you doing?**

Changing my start gcode

**What did you expect to happen?**

That Cura would add my start gcode to the *start* of the exported gcode

**What happened instead?**

Cura added a bunch of other stuff *before* my start gcode which would cause the printer to malfunction.

**Cura version**

2.4.0-Beta

**Notes**

My printer needs M80 to be the very first line to turn on the power supply, thus my start gcode is this: https://gist.github.com/ntoff/699213b39abb74c195385dfc34b07d36 which works in cura v15.x but in 2.4 it adds its own temperature lines before my start code (excerpt from a saved gcode file under 2.4): https://gist.github.com/ntoff/d87b47f6ee1ac6b1845b8031f3926f14

Obviously also the temperature parameters don't seem to be the same as 15.x but the main concern is that M80 is never the first line.

Ghostkeeper, 2017-01-05:
Normally the start g-code has a lot of stuff like priming that needs the nozzle at printing temperature. Not all though.

If you put `M109 S{material_print_temperature}` in the start g-code, or anything with `{material_print_temperature}`, CuraEngine won't put the temperature before it.

So by default it puts the temperature there because it's easy for many printers (e.g. those without any special start g-code). But if your printer requires a special procedure at start you can place the material pre-heat in the start code yourself as well. Anything in brackets gets replaced by the value of the setting between the brackets.

Ghostkeeper, 2017-01-05:
I think your start g-code would become something like this:

    M80 								; turn printer on
    G21        						; metric values
    G91									; Relative Positioning
    M82        						; set extruder to absolute mode
    G1 Z+10							; makes sure the nozzle isn't going to hit the bed when it homes
    G28	Y								; home y
    G90        						; absolute positioning
    G1 Y190							; move bed away from e3d fan of death
    M190 S{material_bed_temperature} 	; heat bed
    M104 S{material_print_temperature}		; start heating
    G28									; home
    M109 S{material_print_temperature} 		; finish heating
    M107       						; start with the fan off
    G92 E0								; reset extruder
    G1 X5 Z0.5 F2500					; move up and across
    G1 Y55 E10 F150					; extrude some material to prime the nozzle
    G92 E0 							; reset extruder
    G1 F{travel_speed}
    ;Put printing message on LCD screen
    M117 Started at {time}

ntoff, 2017-01-06:
>But if your printer requires a special procedure at start you can place the material pre-heat in the start code yourself as well.

Yes, I figured that part out but the issue still stands that I would have assumed *start* gcode would be the absolute first thing added before anything else in there.

Also is there any list of those parameters? I googled for a bit but didn't find anything and just resorted to sifting through the source code to figure out what they were renamed to.

Ghostkeeper, 2017-01-06:
Yes, over here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

They are all keys of all settings.

I wonder, by the way, what would happen if you were to include the `{machine_start_gcode}` key... 😄

ntoff, 2017-01-07:
>I wonder, by the way, what would happen if you were to include the {machine_start_gcode} key...

Don't know but I'm willing to try for science since cura 2.4 isn't my main slicer (cura 15.04 is), I don't mind if I break it.

Ghostkeeper, 2017-05-28:
In short, this is expected behaviour because it's just handy that the temperature gets inserted there, and the print header also gets included before the start g-code because it usually requires information that is unavailable in a start g-code.

closed 2017-05-28 by Ghostkeeper