Ultimaker/Cura #1308: Material diameter not loaded from material profile

dWoolridge, 2017-01-05:
Cura 2.4.0 beta
Windows 7 64-bit

The material diameter shown in the Print Setup panel does not change when loading a new material profile.

To Duplicate:

1. Create a custom material with a diameter of 1.75mm
2. Make sure that the Material->Diameter is checked in "Setting Visibility"
3. Load the default Material "Generic->ABS"
4. Load the default Quality Profile "Normal Quality - 0.1mm
5. Note that the Material->Diameter is shown as 2.85mm
6. Load the custom material with the 1.75mm diameter
7. Note that the Material->Diameter stays 2.85mm 

I have to adjust the flow slightly for each roll of filament that I have to optimize the printing.  I had planned to manage this by creating a Material profile for each roll of filament that I have.

It would also be nice to have a % flow adjustment for each material, rather than that also being saved in the Quality Profile.

Thanks,
--Doug

gregs007, 2017-01-12:
This also happens in 2.3.1 for Mac

morphlaugh, 2017-01-17:
Confirmed-- 2.3.1 for Mac also fails to load the 1.75mm diameter from the material profile when selected.

Ghostkeeper, 2017-01-22:
It seems we have two diameter settings:

* https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L1385
* The diameter in the XML profiles. This is obtained here: https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L426 but that variable is never used after that.

fieldOfView, 2017-01-22:
Seems ```"diameter": "material_diameter"``` needs to be added to ```__material_property_setting_map``` here:
https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L581

fbertram, 2017-02-02:
I might be missing something here, but as most printers will only accept filament of a certain diameter, it might make sense to move the material diameter to the printer profile. The advantage is, that the generic profiles for materials will be independent from the diameter. In case a printer can really load materials with multiple diameters, separate printer profiles would be required, which is still better than duplicating many material profiles.

Ghostkeeper, 2017-02-06:
The material profiles are dependent on the diameter though. A 2.85mm filament will require a bit more heat than a 1.75mm filament.

fieldOfView, 2017-02-06:
I think a good course of action would be to create an additional set of generic_* materials (pla, abs, etc) with a diameter of 1.75mm. The materials view and menus would then be filtered to only show materials which match to the currently selected machine. 1.75 mm machines are not going to be able to print with Ultimaker filaments.

nallath, 2017-04-14:
I've had some lenghty discussions regarding this with a lot of people at Ultimaker.

We think that the best solution is to indeed make material profiles for multiple diameters, but to also give them a different GUID (So generic_pla_285 will be identifiable from generic_pla_175). This is needded because the nfc chips on the material hold GUID, but not the diameter. If we are gonna do this (and we are), the UM3 needs to recognise (and refuse) 1.75 materials.

fieldOfView, 2017-04-14:
Yes, that was my idea all along. The GUID would basically be linked to a "SKU", and a roll of 3mm material is a different product than a roll of the same material in 1.75mm material.

nallath, 2017-04-14:
Well, GUID can't be linked to a SKU, as a roll of 300gram should have the same GUID as a roll of 500gram (but it should be a different SKU).
I simply checked this with everyone to ensure that there is consensus before moving forward. We didn' t really think material profiles would be adopted in the way they are now, so we now have a responsibility to make sure it works for everyone.

closed 2017-11-13 by ChrisTerBeke