Ultimaker/Cura #1309: Remember last file location

claell, 2017-01-05:
I'd like to have an option to either set a default path or to remember the last location from where files were opened.

Ghostkeeper, 2017-01-06:
It should always remember the last location. Is your issue that it doesn't remember the last location or that there is no setting to enable/disable remembering?

Also, what is your operating system? If there is a bug with this, it will be highly dependent on whether you use Windows, OSX or Linux.

claell, 2017-01-06:
For me it does not remember that last location. Personally I don't need to have this as an option.

I am using Ubuntu 16.10 with the AppImage version 2.4.0 BETA.

claell, 2017-01-06:
I just tried again. It seems to remember the location while the app is open, but not after closing and starting again.

Ghostkeeper, 2017-01-22:
I could reproduce this. I filed a bug to get this fixed.

ianpaschal, 2018-02-27:
This is fixed in 3.2.1.

closed 2018-02-27 by ianpaschal